robotology/icub-firmware #10: Clarify diagnostics messages

marcoaccame, 2015-09-25:
Diagnostics messages sent by the EMS / MC4PLUS boards need to be easily understood.

There are several ways do do so:
* in file EoError.c in icub-firmware-shared repository, add a textual description which explains the meaning of the two par16 and par64 parameters.
* increase the level of details of some errors (e.g., the generic motor error), possibly splitting a generic situation into more detailed ones. 
* implement at robotInterface level a parser which expands the compact form of the diagnostics messages into more human readable prints.

marcoaccame, 2015-09-25:
a further requirement is:
* add in repository a document which explains in details the errors so that anyone can easily understand what is happening.

marcoaccame, 2015-10-22:
@ale can begin improving messages related to MC then he can assign back the issue to me so that I can improve other messages.

valegagge, 2015-11-24:
We'll change spike diagnostic: fw sends diagnostic message to robot interface with number of spikes occurred in last second. If there aren't spikes, then no diagnostic message is sent. 

@davege89 implemented this behaviour on branch for test. He will merge it in master.

valegagge, 2015-11-24:
related issue #45

davege89, 2015-11-24:
Done with SHA: 025c24864d3cbb0b492b11c178037431ef5983cb.
@ale-git has worked on others MC diagnostics message with SHA: d8b0cacaed4b5c836bed3454c72e8557a60e80d8

marcoaccame, 2015-11-25:
@ale-git , i have seen that you have changed the description of the errors, but I think it is still required  the description of what par16 and par64 contain. The aim is to print all the information required to understand what is happening and where to search the robot for fixing it without the need to browse the source code.

For instance, the error `eoerror_value_MC_aea_abs_enc_invalid`, the one that @DanielePucci comments about in issue #45 has text:
```
"MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings."
```
but does not have indication of the the joint in which the problem happens (which by now I know it is in par16, but people from P&M do not know). Also, what is inside par64?

DanielePucci, 2015-11-25:
Yes guys, this bad diagnostics make us loos years.

ale-git, 2015-11-25:
I've added the "par16 = ID of joint" specification in all messages since the meaning is always the same, and removed all redundant par64. Anyway, I think that robotInterface should interpret the messages and print them in readable format, instead of printing them in raw format together with the decription book :-)

DanielePucci, 2015-11-25:
@ale-git :+1:

valegagge, 2023-02-22:
A very old issue. If necessary we'll reopen a new one. Closing...

closed 2023-02-22 by valegagge