robotology/icub-firmware #6: check pwm output if joints is close to limit

valegagge, 2015-09-25:
In openloop mode, if joint is on min or max limit joint, the given pwm is ignored so the joint doesn't move.

davege89, 2015-09-29:
When the open loop control mode is active for a joint, the function which returns the PWM to be applied (before the motors decoupling)  simply uses the value provided using the yarpmotorgui, stored inside an internal variable.  However, a check based on the actual position (always updated using the reference encoder) of the joint is done: if the current pos is out or exactly on one of the two boundaries, no PWM is applied.
The snippet which produces this behaviour is here:
https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/board/mc4plus/appl/wip/src/eoappservices/mc/EOaxisController.c#L798

Here I'm referring to the MC4plus local controller (still different from the one used by the EMS).
It is worth asking @ale-git if the updated controller for EMS changes this behaviour or not.
In my opinion, it is safe to stop the movement of the joint when one of the limit is reached, but we should let the user change direction ( = changing the sign of the PWM) so that the joint can return inside the limits without changing the control mode.
A clarification regarding the correct policy is needed.

iron76, 2015-10-05:
@randaz81 should have an opinion on this issue.

davege89, 2015-10-15:
We decided to implement the policy proposed in the previous comments:
the joint should be able to go back when a limit is reached.

I'm going to commit and push the modification in the next few days.

davege89, 2015-10-22:
The fix is working, but @valegagge and I discovered a inherent problem: if the joint doesn't reach the hw limit  with the given PWM (e.g. limit set inside XML is **35** and joint stop moving at **34.9**, because of friction or mechanicals issues), the board does not stop providing PWM to the motor.
This behaviour could lead to the breaking of the motors and can be avoided with different strategies.
We are still thinking about the best solution, which will be discussed during the next meetings. 

A new issue with reference to this one will be created.

commit 9d86a168 (2015-10-16): When the joint is in open-loop control mode, it stops moving when it reaches one of the hw limits, and it can go back changing the sign of the PWM.

Fixes #6
closed 2015-10-22 by davege89