robotology/icub-firmware #8: Migrate FW projects for MAIS, STRAIN, MTB into MPLAB-X

marcoaccame, 2015-09-25:
It is important to migrate to the new tool-chain as experience shows it it much more user-friendly.

marcoaccame, 2015-10-22:
there are two available licences for mplab-x which can be requested to @mbrunettini. @valegagge can ask help to @ale-git who has already the 2foc application on mplab-x

valegagge, 2016-07-27:
I added mplabX project files for strain and mais application on master branch. 
(see 587862a78745c7e3b005a9d89da73b50795e08fa and 70db9e71032f6e08ccc8cbdd9f8158202a2e1081 )

I used MplabX ide v3.35 and compiler Mplab c30 ver 3.31.
Note that the compiler is the same used in mplab, so the executable should be the same.
Before using executable compiled with new ide, i would like test it on a robot, but currently there is no one in lab.

valegagge, 2016-07-27:
Also added Mplabx project files form mtb.

valegagge, 2016-08-31:
I flashed strain, mais and skin firmware compiled with MPLABX ide on purple iCub.
 - Strain version is 3.0.6
 - Mais version is 3.1.3
 - Skin version  is 2.17.1
The executables are here: icub-firmware-build@132f49a78b2a15d01c64706680c629291a77f70c

If in a few days no error occurs, I'll release these new versions.

valegagge, 2016-09-02:
No error occured, so I moved the executables in master branch.

closed 2016-09-02 by valegagge