robotology/icub-firmware #14: Velocity control of arm joints #4-14 does not work

pattacini, 2015-10-02:
As per title, we've just checked with @randaz81 that velocity control for a specific subset of arm joints (0-based) does not work.

pattacini, 2015-10-02:
This applies to `ETH` robots.

marcoaccame, 2015-10-02:
Ok Ugo, thanks for reporting the issue.

As we already discussed together (you, @randaz81 and I), these are the joints whose control is done on the MC4 can boards attached to CAN1 of the EB2 / EB4 ethernet boards.

Moreover, the velocity control on joints [0, 3] works fine. For this first set of joints the control terminates of the ethernet boards EB1 / EB3.

Thus, we suspect that the boards EB2 / EB4 do not propagate the relevant commands to the MC4 boards.

I shall deal with the issue as first thing Monday morning.

marcoaccame, 2015-10-09:
The velocity control on lower arms of `ETH` robots is now working.

@randaz81 and I have worked together to fix the issue.

The problem was that the MC4 boards did not receive a suitable max velocity of joints, thus they could not move. It was necessary to add the parameter in the xml files, to add a field to the config of the joint sent by embObjMotionControl to the EMS boards, and have the EMS send the value to the relevant MC4 board.

In order to work properly, ETH robots need to pull the latest icub-main, icub-firmware-shared, icub-firmware-build and:
* re-build icub-main on PC104
* perform a FW update of the EMS boards.

This operation is already done on iCubGenova02, which is now working properly.

pattacini, 2015-10-11:
Thanks for sorting this out.
Could you update `Bluey` as well please?

closed 2015-10-11 by pattacini

marcoaccame, 2015-10-12:
iCubgenova04 is updated

pattacini, 2015-10-12:
thanks a lot!