robotology/icub-firmware #18: MotionControl: the controller doesn't update joint position if velocity is very high

davege89, 2015-10-15:
@valegagge and I discovered this bug on eyelids joint with out eyelids mounted, so the motor can rotate freely.
If an high speed is set for the current trajectory, encoders values change quickly and the firmware detects these changes as *spikes* and doesn't update joint's position (values are discarded, being considered invalid).
Indeed, robotInterface reports "warning" on that encoder because of these spikes.

We should find a way to:
* find a common value that is reasonably valid for AEA, AMO, Incremental encoders etc...to put the right threshold in order to detect a spike.  Otherwise we shall need a specific parameter for every joint.
* avoid the risk that permanent spikes stop reading of position and thus make motion control unstable. E.g. if spikes stay longer than X ms --> put the joint in HW FAULT

A discussion with @ale-git  is needed in this case.

The issue is related to the bug https://redmine.icub.iit.local/issues/789, which will be closed along with it.

davege89, 2015-10-23:
First attempt to find a solution has been made on branch ``armV3`` with SHA: 9514a2b692931d584fe60c3fdeaff1156911a946.

Basically:
* the threshold has been left static (112 iCubdegrees --> 2/3 of degree), cause it seems consistent even with different encoders
* if spikes are detected for a very long time (~1second) around the same area, the joint is put on HW Fault.

@ale-git, check whether this can be useful for you.

valegagge, 2015-11-13:
moving eyelids joints in position control with high velocity (about 90) thi error happens:
[image]

valegagge, 2015-11-17:
@davege89 and I did some test with a new fw that sends num of occurred position spikes in last second.
We noticed that:
- number of spikes is proportional to velocity of joint
- if joint is idle and we move joint by hand, spikes occur.

The figure below reports position and output during joint is moving.
[image]

davege89, 2015-11-24:
After extensive testing (using also a dedicated branch ``spikes_test`` to produce special FW with enhanced diagnostics), @valegagge and I can state that:
* the AEA encoder which reads the position of the joint is not signalling any errors due to: faults in SPI transmission, parity errors (noise) or wrong exposition to the magnetic field.
* with the help of @maggia80, we reduced the distance between sensor and magnet; this action makes the spikes disappear.

In conclusion, we are oriented to think that big differences in measurements between successive samples (!! e.g. 40deg in 10ms!! ) were caused by the defective relative positioning of sensor and magnet.
The last counterproof could be represented from the voluntary reproduction of the error, putting the two parts in their original position again.

closed 2016-07-06 by valegagge