robotology/icub-firmware #19: MotionControl: the controller limits PWM to 10000

davege89, 2015-10-15:
The controller limits PWM to 10000 (using a MACRO) even if PWM limits (configured using XML files or yarpmotorgui) are bigger than 10000.

Since new_fw_ems has been merged into master, the 2FOC limits the current. Thus we can remove the 10k hidden limit.
For mc4plus, the limit is given by the 3360 of PWM peripheral.

We need to:
* add a method for retrieving max PWM of  the board (= for the type of control used, 2foc-based, mc4plus-based etc...)
* in order to hide hardware details to higher levels, consider a limit of ±32k, normalise the values for all the cases according to their limits.

The issue is related to the bug https://redmine.icub.iit.local/issues/790, which will be closed along with it.

davege89, 2015-10-20:
A first attempt to resolve the issue has been made on branch ```armV3``` with the commit SHA: 850e036ce9579b4da23c65606ed1db8f1590ab8f.

I tested its behaviour on my experimental setup board, and in the next few days I'll try with the HeadV3 and ethernet robots.
So far it seems working properly.

ale-git, 2015-10-22:
We must add a maxpwmofmotor parameter in the .xml files, in the Limits section.
- Marco will add the parameter
- Alessandro will make use of it in the pwm decoupler section

valegagge, 2015-10-22:
What is the meaning of  maxpwmofmotor? the max hardware pwm value or is it a user limit?
Which is the relationship between this new param and maxPwm param in PID configuration in xml file?
maxpwmofmotor value could affect the broadcasted controller output. 

In the commit SHA: 850e036 @davege89 had worked on that ad should be awere of changes.

Maybe it should be better if we talk about this in next firmware meeting.

valegagge, 2016-07-06:
We added Motor pwm limit in xml file. This value limits the pwm given to the motor. (After decoupling)

Currently each board type has a different renge of pwm values: for example mc4plus works in (0, 3360) while 2foc in (0, 32000). Currently user cannot get this information.

closed 2016-07-06 by valegagge