robotology/icub-firmware #21: Currently mc4plus firmware doesn't manage max current of motor.

valegagge, 2015-10-20:
Motor controller engine does't check if current value  exceeds its threshold, so cannot set joints in hardware fault if it happen.

The threshold of current is set up by current param of limit grroup in xml file.
embObjMotionControl module sends current's threshold to firmware, but fw doesn't  update this value because it doesn't implement " set max current of motor" callback function of eth protocol,

davege89, 2015-10-21:
This issue is strongly related to #16 and #17.
My opinion is that a one shot solution to a more general "handling of currents for mc4plus" is needed to resolve these misbehaviours.
I'll check the 2FOC firmware to have an idea on how to do that.

valegagge, 2016-01-28:
Overcurrent fault is checked in mc4plus firmware version 2.21. 
I tested it by settining overloadCurrent=0 in xml config  file.
I not tested by changing overloadcurrent by yarp API, but I suppose it works fine.

closed 2016-01-28 by valegagge