robotology/icub-firmware #30: Need for longer calibration timeout

ale-git, 2015-11-05:
@marcoaccame @davege89 @randaz81 
Some type of actuators need a calibration timeuot longer than 30 seconds. Is it already configurable? Otherwise, we should make it configurable or at least increase it in the firmware...

davege89, 2015-11-05:
IMHO this issue is not firmware-related. I gave a quick look through the code and all the calibration timeouts of the joints are handled by icub-main/src/libraries/icubmod/parametricCalibrator/parametricCalibrator.cpp, used by robotInterface.
They don't seem to be configurable at the moment, but we could add a special case for some new calibration types.

randaz81, 2015-11-20:
We coulf modifiy the parametricCalibratorEth.cpp to set a configurable timeout from the xml file. I'll give a look.