robotology/icub-firmware #34: Increase the status of joints

marcoaccame, 2015-11-18:
There is the need to increase / reorganize the status of joints. in particular, it is required to add some entries in data structures related to the status of PIDs.

marcoaccame, 2015-11-18:
In a recent technical meeting we have agreed to differentiate the status of the PID used to control the joint according to the control/interaction mode in use.

The resulting union `eOmc_joint_status_ofpid_t` is the following.

```C

/** @typedef    typedef struct eOmc_status_ofpid_generic_t
    @brief      contains the status of a generic PID 
 **/
typedef struct
{
    int32_t                     reference1;     /**< the */
    int32_t                     reference2;     /**< the */
    int32_t                     error1;         /**< the */
    int32_t                     error2;         /**< the */
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_generic_t;  EO_VERIFYsizeof(eOmc_status_ofpid_generic_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_olo_t
    @brief      contains the status of an openloop PID 
 **/
typedef struct
{
    int32_t                     refolo;         /**< the */
    int32_t                     dummyref2;
    int32_t                     dummyerr1;
    int32_t                     dummyerr2;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_olo_t;      EO_VERIFYsizeof(eOmc_status_ofpid_olo_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_pos_t
    @brief      contains the status of a position PID 
 **/
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     dummyref2;      /**< the */
    int32_t                     errpos;         /**< the error */ 
    int32_t                     dummyerr2;
    int32_t                     output;         /**< the output */    
} eOmc_status_ofpid_pos_t;      EO_VERIFYsizeof(eOmc_status_ofpid_pos_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_trq_t
    @brief      contains the status of a torque PID 
 **/
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     reftrq;         /**< the */
    int32_t                     errpos;         /**< the error */ 
    int32_t                     errtrq;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_trq_t;      EO_VERIFYsizeof(eOmc_status_ofpid_trq_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_trq_t
    @brief      contains the status of a torque PID 
 **/
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     reftrq;         /**< the */
    int32_t                     errpos;          /**< the error */ 
    int32_t                     errtrq;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_trq_t;      EO_VERIFYsizeof(eOmc_status_ofpid_trq_t, 20);

/** @typedef    typedef uinion eOmc_joint_status_ofpid_t
    @brief      eOmc_joint_status_ofpid_t contains the status of a PID.
 **/
typedef union                  
{   
    eOmc_status_ofpid_generic_t     generic;
    eOmc_status_ofpid_olo_t         openloop;
    eOmc_status_ofpid_pos_t         stiffpos;
    eOmc_status_ofpid_pos_t         complpos;
    eOmc_status_ofpid_trq_t         torque;
} eOmc_joint_status_ofpid_t;        EO_VERIFYsizeof(eOmc_joint_status_ofpid_t, 20);

```

marcoaccame, 2015-11-18:
I have implemented the changes and now I need to test them on the iCubGenova02 robot.

marcoaccame, 2015-11-19:
I am experiencing doubts on how to map the old legacy pid variables onto the new ones.

Thus, I have decided to add also a legacy field which can be used just now everywhere in the code. Later on we can use the new fields and drop the legacy ones.

We can also review the content of new unions more easily

```C

/** @typedef    typedef struct eOmc_status_ofpid_legacy_t
    @brief      contains the status of a generic PID as until 19 nov 2015
 **/
typedef struct
{
    int32_t                 positionreference;  /**< the reference of the position pid */
    int32_t                 torquereference;    /**< the reference of the torque pid */
    int32_t                 error;              /**< the error of the pid */ 
    int32_t                 output;             /**< the output of the pid */ 
    int32_t                 filler;
} eOmc_status_ofpid_legacy_t;  EO_VERIFYsizeof(eOmc_status_ofpid_legacy_t, 20);

/** @typedef    typedef uinion eOmc_joint_status_ofpid_t
    @brief      eOmc_joint_status_ofpid_t contains the status of a PID.
 **/
typedef union                  
{   
    eOmc_status_ofpid_legacy_t      legacy;
    eOmc_status_ofpid_generic_t     generic;
    eOmc_status_ofpid_olo_t         openloop;
    eOmc_status_ofpid_pos_t         stiffpos;
    eOmc_status_ofpid_pos_t         complpos;
    eOmc_status_ofpid_trq_t         torque;
} eOmc_joint_status_ofpid_t;        EO_VERIFYsizeof(eOmc_joint_status_ofpid_t, 20);

```

valegagge, 2015-11-19:
I have a doubt: when a joint is controlled in compliant position, the status of pid should contain torque reference and torque output also?

In other words, should we defined following struct?
```
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     reftrq;         /**< the */
    int32_t                     errpos;         /**< the error */ 
    int32_t                     errtrq;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_compPos_t;      EO_VERIFYsizeof(eOmc_status_ofpid_compPos_t, 20);

/** @typedef    typedef uinion eOmc_joint_status_ofpid_t
    @brief      eOmc_joint_status_ofpid_t contains the status of a PID.
 **/
typedef union                  
{   
    eOmc_status_ofpid_generic_t     generic;
    eOmc_status_ofpid_olo_t            openloop;
    eOmc_status_ofpid_pos_t           stiffpos;
    eOmc_status_ofpid_compPos_t  complpos;
    eOmc_status_ofpid_trq_t             torque;
} eOmc_joint_status_ofpid_t;        EO_VERIFYsizeof(eOmc_joint_status_ofpid_t, 20);
```

randaz81, 2015-11-19:
yes, with output = output of the inner torque pid loop

valegagge, 2015-11-19:
In short new structures are:

```C

/** @typedef    typedef struct eOmc_status_ofpid_generic_t
    @brief      contains the status of a generic PID 
 **/
typedef struct
{
    int32_t                     reference1;     /**< the */
    int32_t                     reference2;     /**< the */
    int32_t                     error1;         /**< the */
    int32_t                     error2;         /**< the */
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_generic_t;  EO_VERIFYsizeof(eOmc_status_ofpid_generic_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_olo_t
    @brief      contains the status of an openloop PID 
 **/
typedef struct
{
    int32_t                     refolo;         /**< the */
    int32_t                     dummyref2;
    int32_t                     dummyerr1;
    int32_t                     dummyerr2;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_olo_t;      EO_VERIFYsizeof(eOmc_status_ofpid_olo_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_pos_t
    @brief      contains the status of a position PID 
 **/
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     dummyref2;      /**< the */
    int32_t                     errpos;         /**< the error */ 
    int32_t                     dummyerr2;
    int32_t                     output;         /**< the output */    
} eOmc_status_ofpid_pos_t;      EO_VERIFYsizeof(eOmc_status_ofpid_pos_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_trq_t
    @brief      contains the status of a torque PID 
 **/
typedef struct
{
    int32_t                     dummyref1;       
    int32_t                     reftrq;                 /**< the */
    int32_t                    dummyerr1;         /**< the error */ 
    int32_t                     errtrq;
    int32_t                     output;         /**< the output */  
} eOmc_status_ofpid_trq_t;      EO_VERIFYsizeof(eOmc_status_ofpid_trq_t, 20);

/** @typedef    typedef struct eOmc_status_ofpid_pos_t
    @brief      contains the status of a compliant position PID 
 **/
typedef struct
{
    int32_t                     refpos;         /**< the */
    int32_t                     reftrq;         /**< the */
    int32_t                     errpos;         /**< the error */ 
    int32_t                     errtrq;
    int32_t                     output;         /**< the output of inner torque pid pos*/  
} eOmc_status_ofpid_compPos_t;      EO_VERIFYsizeof(eOmc_status_ofpid_compPos_t, 20);

/** @typedef    typedef uinion eOmc_joint_status_ofpid_t
    @brief      eOmc_joint_status_ofpid_t contains the status of a PID.
 **/
typedef union                  
{   
    eOmc_status_ofpid_generic_t     generic;
    eOmc_status_ofpid_olo_t            openloop;
    eOmc_status_ofpid_pos_t           stiffpos;
    eOmc_status_ofpid_compPos_t  complpos;
    eOmc_status_ofpid_trq_t             torque;
} eOmc_joint_status_ofpid_t;        EO_VERIFYsizeof(eOmc_joint_status_ofpid_t, 20);

```

marcoaccame, 2015-11-19:
corrected the above data structures in branch bigger_mc_status.
tested correct behavior of robot iCubGenova02 by using the `legacy` union.
done merge on master.
released a new FW version 2.17.

now there will be a new issue opened, aimed at using in FW and in embObjMotionControl.cpp the correct fields of `openloop`, `stiffpos`, `complpos`, `torque`.

closed 2015-11-19 by marcoaccame