robotology/icub-firmware #36: [ETH] Error when opening boards associatd with strain

DanielePucci, 2015-11-19:
I am currently using the `iCubGenova02` (ETH robot). Then I open the `canLoader` with the `--calib` option, i.e.
~~~
 canLoader --calib
~~~

When I click on the `Calibrate` button after having connected to  a board associated with a strain,e.g. 

[image]

I get the following error on the terminal:

~~~
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
[ERROR]strain_get_matrix_rc : invalid response
~~~

@valegagge @randaz81

valegagge, 2015-11-27:
The problem was due to a wrong commit: on master branch there was implemented a new feature for use multiple calib matrices that needs a particular version of canLoader.

Moreover I noticed  bug fix about set_curr_tare message was missing.

Therefore I moved implementation of the new feature in a new branch called strain_with_ABCmatrix and restored bug fix on master. 

I released a new version of strain fw 3.0.4 in icub-firmware-build.

I tested it on my strain and all is ok.

@DanielePucci could you verify if the problem is solved? thank you

DanielePucci, 2015-11-27:
Thank you very much @valegagge. I think that, however, we freeze the code on the iCubGenova02, and test this new feature as soon as the robot comes back (Monday 7th). Can't we test this improvement on the iCubGenova04?

valegagge, 2015-11-30:
yes, of course!
I'll wait your feedback

valegagge, 2016-02-15:
I updated strain sensors on iCubGenova04 with 3.0.4 version. All is ok.

closed 2016-02-15 by valegagge