robotology/icub-firmware #38: [ETH] Add an option to the canLoader to change the status of the boards into "gateway"

DanielePucci, 2015-11-19:
Before addressing a board through the `canLoader`, e.g. that of a strain, one has to open the `ethLoader` and click on the button `Discover` to change the status of the boards into `CAN gateway`, which allows the users to use the `canLoader`. 

It could be awesome if we could avoid this intermediate step in one way or another. An option is to add the `Discover` button into the `canLoader`.

@valegagge @marcoaccame @davege89 @randaz81

DanielePucci, 2015-11-19:
Another option is that the `canLoader` automatically runs a `Discover` when clocking on `Connect`

randaz81, 2015-11-20:
The feature you requested cannot be added in the canLoader tool.
However, both CanLoader and EthLoader have to be rebuilt ex novo, basically because we are deprecating GTK, so I'd like to put this feature in the new version of the tool (only one tool that includes the other two).

closed 2015-11-20 by randaz81