robotology/icub-firmware #47: torque at 32 bit resolution

marcoaccame, 2015-11-23:
@DanielePucci has requested to have torque represented w/ 32 bit resolution instead of as a int16_t with 1 milliNm per unit so that we can manage torques of more than 32 Nm.

marcoaccame, 2015-11-23:
done `torque32bits` branches in icub-main, icub-firmware-shared, icub-firmware. added in `master` a icub-firmware-buld/testing/ems-torque32bits.hex w/ version 222.20

tested on purple robot. basic movement seems to be working. 

@DanielePucci can you please verify it it is OK also the torque control?

ale-git, 2015-11-23:
@marcoaccame In the axisController the torques are already represented with int32, but the torque measure setting function eo_axisController_SetTorque was still 16 bit, did you change it?

DanielePucci, 2015-11-23:
As we discussed today, further investigations deserved to be carried on to clarify the soundness of measured and reference torques coming from robotInterface. Then, tests on the new representations will also be preformed. @francesco-romano @traversaro

marcoaccame, 2015-11-24:
Yes @ale-git, I have changed it into:
```C
extern void eo_axisController_SetTorque(EOaxisController *o, eOmeas_torque_t trq)
```

marcoaccame, 2015-11-25:
tested again on iCubGenova02 and merged into master.

closed 2015-11-25 by marcoaccame