robotology/icub-firmware #56: PositionMove with RefVel = 0

randaz81, 2016-01-26:
PositionMove commands with reference velocity = 0 should be not executed by the ETH firmware (while the reference position value should be updated instead).
Currently the joint performs a jerky and dangerous movement (check for possible divisions by zero). 
To be verified in EMS, MC4PLUS and MC2PLUS firmwares.

ale-git, 2016-01-27:
@randaz81, @valegagge 
It is not a bug: I remember that it was decided in a software meeting some years ago that velocity=0 in positionMove means "maximum velocity". The function involved is 
extern void eo_trajectory_SetPosReference(EOtrajectory *o, int32_t x0, int32_t xStar, int32_t velAvg)
and specifically
if (!velAvg) velAvg = o->vel_max; // line 163  of EOtrajectory.c 
If we need to change this counterintuitive behaviour, I think that a position command with velocity=0 should be simply rejected, because it means "go to X by staying still (lol)"

ale-git, 2016-01-27:
If I remember correctly, rejecting the position commands with velocity=0 was the original behaviour of the function.

lornat75, 2016-01-27:
I agree that the current behavior (max velocity) is weird, I think it is not too bad if the joint does not move. If I recall correctly this was the behavor of industrial motion control boards I used in the past.

ale-git, 2016-01-27:
Now positionMove with refVel = 0 is rejected (but position reference is stored). 
velocityMove with refAcc = 0 is interpreted as a velocityStop, because simply rejecting the command would mean that the joint continues to move with the previous velocity, since the velocity timeout watchdog is reset when a new velocity command arrives. Otherwise, a continuous stream of velocityMove with refAcc = 0 would force the joint to move at the last valid velocity setpoint without any watchdog expiring.

closed 2016-01-27 by ale-git