robotology/icub-firmware #66: Improve check of torque valid value in "ethernet motion control" module.

valegagge, 2016-05-10:
Currently, the motion control module, if receives the same value for 100 ms, puts motors in fault because thinks the virtual strain has stopped to trasmit.

In case strain is saturating, it sends the last valid value.
So, motor controller interprets the saturation of strain like no trasmission and puts the motor in fault. 

Therefore, it is necessary to improve this check in order to continue to control motors even if strain is saturating.

valegagge, 2016-05-20:
@ale-git modified check with this commit d6cb08c77c837c095757aec828518debc05674f1
Now I'm going to add watchdod on receiving can messages. (Same watchdog can be used to check mais and skin messages.)

randaz81, 2016-05-20:
100ms are already enough.
Better to remove the *2 from:
 WatchDog_set_base_time_msec(&o->trq_fbk_wdog, 2*DEFAULT_WATCHDOG_TIME_MSEC);

randaz81, 2016-05-20:
> In case strain is saturating, it sends the last valid value.
So, motor controller interprets the saturation of strain like no trasmission and puts the motor in fault.

The controller does not uses directly the strain values, it uses the torques estimated by wholeBodyDynamics. This means that wholeBodyDynamics is sending the same value multiple times... which is very strange and should be investigated (even assuming that the strain is saturating, estimated torques should be always slightly different because the encoder and the IMU values are not the same)

traversaro, 2016-05-20:
Yep, I can confirm that this quite strange, as the value should always be changing (at least an epsilon) in all existing version of `wholeBodyDynamics(Tree)`. @valegagge did you experience this in a real robot/setup?

valegagge, 2016-05-20:
I know motorcontroller doesn't use strain value, but the torques estimated by wholeBodyDynamics.

The imlplemented check behaves as fallowing:
put motor in hwfault if receives the same value for 50 times. 

During test some joints went in fault, So this mean that motor controller receives the same value for 50 times.

Currently I have not enough time to study in deep this issue, but after rapid analisys, I suppose:
1 ) wholeBodyDynamics does't send torque value every millisec, but evry 5 or 10
2) 5 or more torque value are quite simililar( differ one from another of small epsilon)
3) this epsilon is lost when torque values are send to ems boards.

Please, give me some time to check my assumptions.
This is a working in progress task. :)

traversaro, 2016-05-20:
The default period of `wholeBodyDynamics` is indeed 10 milliseconds, and I suspect that the virtual analog sensor in the `embObjMotionControl` has a similar default period. My feeling is that for those 50 ms for a combination of reason (delays in network/thread scheduling?) no estimate coming from WBD is arriving to the the motorcontroller. 
Probably this is correlated in some ways with the STRAIN saturating, but I don't know the reason behind such a correlation.

valegagge, 2016-05-20:
@traversaro I think my first assumption that relates strain saturation with this issue was not correct.

If the tx period of wholeBodyDynamics is 10 ms and if five consegutive torque values are quite similar then it is very likely this problem happens.

traversaro, 2016-05-20:
Yes, I was not considering the precision lost in double ---> EMS unit conversions. What is order of magnitude (in Nm) of the lost precision?

randaz81, 2016-05-20:
well, depends on the fullscales you set in the configuration files.
Assuming a fullscale of +-25 Nm, the minimum torque value which can be represented is 25/32768=0.76 mNm (which is quite a small number...)

traversaro, 2016-05-20:
Definitely. I then have the strong feeling that no torque estimate is received for 50 ms.

valegagge, 2016-06-01:
After studing in deep, I get following conclusions:

1) WBD sends data every 10 ms (it uses a yarp rate thread)
2) Virtual wrapper reads WBD data in blocking way. When it receives something, calls embObjMotioncontrol::UpdatedMeasures function
3) embObjMotioncontrol::UpdatedMeasures limits input torque to "TorqueMax" value (param in XML file) and sends limited torque value transformend in milli Nm to firmwarwe.
4) In case of strain saturation, ems notifies  the user by warning messages.
5) In case of saturation of WBD values, embObjMotionControl notifies  the user by warning messages.

Speaking with @traversaro and @ale-git , the expected behaviour of firmware controller is as following:
a joint controlled in torque switches to hw fault state if the controller doesn't receive torque value for 100 ms. The motor controller of firmware should not enter into the merits of torque value validity.
High level applications will be veify to receive updated values from strain.

In light of these facts, in firmware I'll remove check on torque value validity and I'll leave watchdog on reception of torque values.

DanielePucci, 2016-06-01:
I'm concerned with point 4): shouldn't it be an error message instead of warning message?

valegagge, 2016-06-01:
ops... yes is an error message

DanielePucci, 2016-06-01:
👍

valegagge, 2016-06-01:
I done changes explained above in 5b7fbdc95659a52c56047b60a3f736f34f33bcb9

closed 2016-06-01 by valegagge