robotology/icub-firmware #67: Check if Coulomb friction compensation on torque control is active

francesco-romano, 2016-05-17:
@DanielePucci and me were trying the identified motor parameters on iCubDarmstadt01 and we noticed that, even changing the coulomb friction compensation parameters, we did not notice any appreciable difference in the behaviour of the torque-controlled joints.

Is the coulomb friction actually considered in the torque control in the firmware?

DanielePucci, 2016-05-17:
@randaz81

randaz81, 2016-05-17:
I'll check it. Just for confirmation, you are speaking about `stictionUp` `stictionDown` parameters, am I right?

francesco-romano, 2016-05-17:
Yes
`stictionUp` and `stictionDwn`

randaz81, 2016-05-18:
In the current firmware build stictionUp stictionDown are unused :(

traversaro, 2016-05-18:
cc @TeoKa

francesco-romano, 2016-05-18:
This explains the behaviour we noticed!
Thanks @randaz81

traversaro, 2016-05-23:
I tried to do some naive investigation, and I noticed two interesting thing: 
* There is a code path in which the coulomb friction compensation is commented out : 
 https://github.com/robotology/icub-firmware/blob/9de4f7c2bb2c6ae4bdc42fc92d8063c51cbd9a78/emBODY/eBcode/arch-arm/board/mc4plus/appl/wip/src/eoappservices/mc.old/EOpid.c#L337 
* There is another code path in which the stiction up and stiction down parameters are (apparently) inverted: 
https://github.com/robotology/icub-firmware/blob/9de4f7c2bb2c6ae4bdc42fc92d8063c51cbd9a78/emBODY/eBcode/arch-arm/board/mc4plus/appl/wip/src/eoappservices/mc.old/EOpid.c#L125

Both this lines of codes are in the `mc.old` directory, so I don't know if they are still actually used or not. 
@francesco-romano @TeoKa @DanielePucci

traversaro, 2016-05-23:
Upon a quick search, I think that Coulomb compensation was committed commented out in https://github.com/robotology/icub-firmware-shared/commit/bfa54caf76734483efbd9a9930ce90bf954b0351 . At some point the source code was moved to the `icub-firmware` repository, so it is not easy to follow the history.

traversaro, 2016-06-06:
Caveat: the new firmware seems to be doing friction compensation in https://github.com/robotology/icub-firmware/blob/0c582590f99b277d3f7e5cba27058a83d1595b33/emBODY/eBcode/arch-arm/embobj/plus/mc/Pid.c#L131 .

ale-git, 2016-06-06:
Fixed by https://github.com/robotology/icub-firmware/commit/e99a6ed4cb6af0da2b9d242aa8793babf31e8d36

traversaro, 2017-07-12:
For reference, the code that compute the torque control PID action is: 
* https://github.com/robotology/icub-firmware/blame/f5850b0d935e814eab5f1a448a6471b141dfcc5f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L579 for coupled joints 
* https://github.com/robotology/icub-firmware/blame/f5850b0d935e814eab5f1a448a6471b141dfcc5f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L583 for not coupled joints 

Both points call the `Motor_do_trq_control` function, that is defined in https://github.com/robotology/icub-firmware/blob/f5850b0d935e814eab5f1a448a6471b141dfcc5f/emBODY/eBcode/arch-arm/embobj/plus/mc/Motor.c#L774 and calls itself the `PID_do_friction_comp` https://github.com/robotology/icub-firmware/blob/06ddcaa3bef996e5b54319759f13ed05b0a75330/emBODY/eBcode/arch-arm/embobj/plus/mc/Pid.c#L147 . For looking at this code, I think it is safe to assert  that **no coulomb friction compensation** is implemented in the low level torque control loop. 

cc @diegoferigo @nunoguedelha @s-dafarra @DanielePucci

francesco-romano, 2017-07-12:
This is consistent with what I observed on the real robot

traversaro, 2021-12-16:
cc @isorrentino @DanielePucci 

I guess it could be a good idea to fix this in this decade. :D

traversaro, 2021-12-16:
By the way, I think we can easily close the issue with the answer:
**no, the coulomb friction compensation is not active**. I will open a separate issue for actually implementing this.

closed 2021-12-16 by traversaro