robotology/icub-firmware #68: STRAIN boards should be checked for 0xFFFF elements in the calibration matrix

traversaro, 2016-06-07:
For unknown reason, it is possible that some values of the calibration matrix turn in `0xFFFF`. 
We should have a check somewhere (the STRAIN board, or the EMS or the embObjAnalogSensor device) to check that no element of the gain matrix is set to `0xFFFF` at configuration time.

traversaro, 2016-06-07:
See https://github.com/robotology/icub-support/issues/236 .

DanielePucci, 2016-06-08:
@marcoaccame @valegagge

randaz81, 2016-06-09:
> We should have a check somewhere...

it should be checked in the STRAIN board firmware.
After saving to the eeprom, we must re-read values and perform a check if values are identical to the ones currently stored in ram.

valegagge, 2016-06-15:
Three questions to understand better the problem:
1) Can you tel me strain firmware version number?
2) When the problem happen, does the canLoader print something on shell?
3) When the problem happen, can you close and reopen the calibration windows and check if the same element of matrix has value 0xffff?

Thank you!

traversaro, 2016-06-15:
Answers relative to the instance of the problem occurred in the `iCubDarmstadt01` (see https://github.com/robotology/icub-support/issues/236).

1) As far as I remember we flashed the latest firmware available at the beginning of May. Probably the 0xFFFF have been there for a long time unnoticed, so it could be possible that the problem was relative to an old firmware. 

2) No part of the software (including the `canLoader` or any driver running on the `yarprobotinterface`) complains at all. We were just extremely lucky because the element that turned in the 0xFFFF (i.e. ~ -1 ) was close to 1, and it was the biggest element of the line. So at the high level we saw that the measured force relative to that line appeared to be in the opposite direction with respect to what we expected. If this problem happened on another element of the calibration matrix, we would probably have failed to detect the problem, and the F/T measure would simply be corrupted in a much more subtle way. 

3) Yes, the `0xFFFF` value persisted even by powering off the sensor. The only way to restore the right value was to flash the original calibration matrix.

valegagge, 2016-06-15:
ok, thanks!

valegagge, 2016-06-17:
@randaz81 if we implement this check, in case of failure what does the strain do?
1) sends an error message
2) retries to write with a max num of attempts
3) 1) and 2)

If it sends an error message can Loader should be able to manage it opportunity.
If I remember correctly, no error messages are provided by CANprotocol.

I think we can add this check in strain firmware, but we also could be make CanLoader application more robust: after every set command, it should read back and check if received data is equal to sent one. We also need to add a canMassage to ask to strain to reload data from eeprom.
Moreover we can split data read periodically (currently almost all) in two set: one made up of changing data (adc values for example) and one from made up of configured data.

In this way, the user should be sure that every sent command has been managed by strain in proper way and the data displayed in GUI are equal to the data stored in strain. We reduce data rate also...

valegagge, 2016-06-22:
In order to close this bug quikly, I decided to implement the check of writings in eeprom in strain firmware and to postpone the canLoader improvements.

In particular:
1)  Now when strain receives SAVE_IN_EEPROM message, it writes data in eeprom then reads them again and compares read data with written data. Sends a rasponse with the result of comparison.

2) CanLoader checks response to SAVE_IN_EEPROM command and prints a message on bash

The changes are in branches named 'canLoaderCheckEeprom'. The new exe ver 3.0.5 is in icub-firmware-build repository in a branch called with same name.

I verified firmware with strain on my test bench. It is ok.

@traversaro . If I remember correctly, there is a test to check the read data are significant, right?

valegagge, 2016-06-22:
yes! raw data read by sensor

traversaro, 2016-06-22:
I think I am missing the question, we can discuss face 2 face tomorrow.

valegagge, 2016-07-06:
Tested with @traversaro . all is ok. 
Released strain version 3.0.5

closed 2016-07-06 by valegagge