robotology/icub-firmware #70: getMotorsPwm of controlBoardDumper  streames PWMs without signs

DanielePucci, 2016-06-13:
I am trying to read the pwms from the ETH robot iCubGenova05. More precisely, I'm using the `controlBoardDumper` launched as follows: 

~~~
controlBoardDumper --robot icub --part left_leg --rate 10  --joints "(0 1 2 3 4 5)" --dataToDump "(getMotorsPwm  getOutputs)"
~~~
and when I read the two ports 
~~~
controlBoardDumper/left_leg/getMotorsPwm
~~~
and 
~~~
/controlBoardDumper/left_leg/getOutputs
~~~
the former streams pwms without signs, and the latter with signs. As for an examples, typical values obtained from `controlBoardDumper/left_leg/getMotorsPwm` of the `left_ankle_pitch` joint are:

~~~
14266.0 
14274.0 
14288.0 
14312.0 
14337.0
14367.0 
14391.0 
  10.0 
  29.0 
  47.0 
  73.0 
 100.0 
~~~
while  typical values obtained from `controlBoardDumper/left_leg/getOutputs` are
~~~
-382.0 
 -382.0
-382.0 
-382.0 
-189.0
-189.0 
-189.0 
3.0 
3.0 
3.0 
3.0 
197.0 
197.0 
197.0 
390.0 
390.0 
390.0 
~~~

@randaz81 @ale-git @marcoaccame @valegagge

DanielePucci, 2016-06-13:
@maggia80

ale-git, 2016-06-16:
The 2FOC fw wasn't upgraded to the latest version, so it wasn't sending the pwm data to the EMS (what you was seeing was the motor rotor position :p ). Fixed.

closed 2016-06-16 by ale-git