robotology/icub-firmware #74: Hardware Fault when faulting the robot

francesco-romano, 2016-07-15:
When the robot is torque controlled and it is faulted, when it is unfaulted the joints goes into HW FAULT instead of idle, and they must be idled one by one.

@ale-git

francesco-romano, 2017-10-27:
@marcoaccame @ale-git this issue is still a problem.

cc @DanielePucci

DanielePucci, 2017-10-29:
Yes, it slows down the robot experimental activities. May we do something about this?

@ale-git @marcoaccame

barbalberto, 2017-10-30:
On the yarpMotorGui there is a button to idle/run all joints of the robot in a single click, is it not enough?

marcoaccame, 2017-10-30:
Hi @francesco-romano and @DanielePucci,

The firmware supports commands to send joints in IDLE/RUN. And I believe that the **yarpmotorgui** already has buttons which sends in IDLE/RUN the body part or the whole robot. 

... @barbalberto just confirmed that.

traversaro, 2017-10-30:
Only the single joint idle button sends [1] the `VOCAB_CM_FORCE_IDLE` command necessary to switch the hardware fault joints from `VOCAB_CM_HW_FAULT` to `VOCAB_CM_IDLE`, see  [2, Section 3.1 ] . This is the reason why on hardware fault you need to put in idle the joints one by one, and the single-click idle/run all joints is not useful.  

[1] : https://github.com/robotology/yarp/blob/f158a2436c8897a81ba599071f3c5f9c366f8d7e/src/yarpmotorgui/partitem.cpp#L813
[2] : http://wiki.icub.org/images/c/cf/ICub_Control_Modes_1_1.pdf

barbalberto, 2017-10-30:
Ok, now I get it. Actually I think it is a bit confusing that idles buttons do different things... I always forget about it.
I'd say the firmware side is doing its job, the issue is that there is no handy way to send `FORCE_IDLE` command to all joints.
Shall we add a separated button to the yarpMotorGui and clarify the behaviour?
@randaz81 what do you think?
Other ideas?

traversaro, 2017-10-30:
Probably there was an implicit point in the issue: until some time ago, after pressing the "big red button AKA fault button" the control mode of the joints was switched to `VOCAB_CM_IDLE`, not `VOCAB_CM_HW_FAULT`. From the perspective of working on the robot, that behavior was much more convenient.

ale-git, 2017-10-30:
Ok I'm going to check the fw - the state should be IDLE actually.

DanielePucci, 2017-10-30:
Thanks @traversaro for clarifying. Leaving aside the (annoying :) ) joint-by-joint idling after faulting the robot, I was wondering why pressing the robot's fault button caused the joints to switch into  `VOCAB_CM_HW_FAULT`. Sorry for not clearing mentioning the two separate points of this issue

valegagge, 2017-10-30:
Hi all,
 @ale-git and I performed following test on green robot:

**TEST 1:** 
1. launched yarprobotinterface, when robot is ready all joints are in position control mode.
2. pressed fault botton and all joints went in idle mode
3. released fault botton and pressed icon "run all parts" of yarpmotorgui: all joint went to position control mode.

**TEST 2:**
1. launched yarprobotinterface, when robot is ready all joints are in position control mode.
2. put a subset of joints in torque control mode
3. pressed fault botton and all joints went in idle mode
4. released fault botton and pressed icon "run all parts" of yarpmotorgui: all joints went to position control mode.

**TEST 3:**
1. launched yarprobotinterface, when robot is ready all joints are in position control mode.
2. put a subset of joints in torque control mode
3. pressed fault botton and all joints went in idle mode
4. released fault botton and put a subset of joints in torque control mode. These joints go in torque control.

These tests show that robot's behaviour is the expected.

Could you show me or @ale-git or @marcoaccame when happen the  issue you refer to, please?

Thank you.
Vale

traversaro, 2017-10-30:
cc @S-Dafarra @gabrielenava

DanielePucci, 2017-10-30:
Hi @valegagge,
thank you very much for further investigating this.

I fear that the number of joints in torque mode has an influence on the obtained robot behaviour

Just as a curiosity, how many joints did you set into torque control mode during your tests?

valegagge, 2017-10-31:
I don't think the number of joints in torque control influences the behaviour of robot. Anyway i set 4 joints of leg in control mode.  

Please, let me know if the robot has a different behaviour,  putting more joints in torque control mode.

S-Dafarra, 2017-10-31:
I was able to reproduce the issue with a simple test. Indeed it seems that the problem is happening when an external controller is sending references to the robot. When pressing the fault we are used to close also these external controllers, which on exit, try to change the control mode to ``POSITION``. From this simple test, it seems to me that this attempt of changing the control mode while faulted causes the boards to go in a hardware fault status, once the fault is released. This also explains why we noticed it when controlling the robot in any control mode different from ``POSITION``.
It can be reproduced also using the ``yarpmotorgui``:
- press the fault without releasing it
- pick any joint and set any control mode through the drop menu
- release the fault.

valegagge, 2017-10-31:
@S-Dafarra please could you attach the log file? maybe there there are some information that let us to understand the behaviour.
thanks
Vale

DanielePucci, 2017-10-31:
@valegagge @marcoaccame @ale-git 
As a [red team](https://en.wikipedia.org/wiki/Red_team), probably it would be better that the firmware group reproduces the experiment performed by @S-Dafarra, and then investigate the problem on-site

francesco-romano, 2017-11-07:
@valegagge @marcoaccame @ale-git 
Any update on this?

S-Dafarra, 2017-11-07:
Here some log files.
I followed the following steps:
- Start the ``yarprobotinterface``
- Open the ``yarpmotorgui``
- Fault the robot
- Set ``Position`` mode from the ``yarpmotorgui`` to the ``l_hip_pitch`` joint
- Release the fault (the ``l_hip_pitch`` joint went in hardware fault)
- Closed the ``yarprobotinterface``

I exported both the entries in the ``yarplogger`` relative to the ``yarprobotinterface``. They are from a **SINGLE** experiment: the ``yarprobotinterface`` create two ``yarplogger`` entries every time it is launched. 

[log_icub-head_yarprobotinterface_3293.txt](https://github.com/robotology/icub-firmware/files/1450005/log_icub-head_yarprobotinterface_3293.txt)
[log_icub-head_yarprobotinterface_3294.txt](https://github.com/robotology/icub-firmware/files/1450006/log_icub-head_yarprobotinterface_3294.txt)

randaz81, 2017-11-08:
1. Only VOCAB_CM_FORCE_IDLE can exit from VOCAB_CM_HW_FAULT and go to VOCAB_CM_IDLE. 
Yarpmotorgui does not currently provide any button to perform a VOCAB_CM_FORCE_IDLE on all joints. This was done for safety reasons, to ensure user to check the reason of the fault before continuing to use the robot. However, a new button can be added... (and, yes, I can change button label to FORCE_IDLE to avoid confusion)
2. **It is not allowed to change control mode during VOCAB_CM_HW_FAULT or VOCAB_CM_IDLE. Firmware should simply reject the command. If this is not the case, this is a bug** @ale-git @valegagge 
3. Joint should not go in VOCAB_CM_HW_FAULT when the fault button is released and control mode is changed, that's a bug. But given 2) stated above, this should never happen.
4. Given 2), when your module quits and tries to change the control mode, nothing should happen. However, it would be also nice to check the status of joint with a getControlMode before attempting to change it. (this is a good practice to do in any case).

ale-git, 2017-11-08:
The hardware fault should be solved in the latest release of the EMS and MC4plus firmware. The bug caused a hardware fault (motor boards in wrong control state) when a set control mode was sent to the boards while in external fault mode.

ale-git, 2017-11-08:
@randaz81 yes, the red button sets the robot in VOCAB_CM_IDLE, not in VOCAB_CM_HW_FAULT.

marcoaccame, 2017-11-08:
@ale-git has tested and released new FW for boards ems and mc4plus (versions v3.13 and v3.10 respectively) on the devel branch of icub-firmware-build. The FW solves the bug.

Please be sure to run the latest FW versions.

closed 2019-08-08 by marcoaccame