robotology/icub-firmware #75: errors when attempting to control the arms with direct position control

serena-ivaldi, 2016-07-28:
Dear all

we are having problems with the direct position control of iCubNancy01. We are trying to control the torso and the arms with a direct position control mode. It works " partially"  (i.e. not for all the joints, which we cannot explain differently, the wrist pronosupination appears in direct control mode in the yarpmotorgui but does not move at all). But most importantly the robotinterface is showing tons and tons of errors of all types.
In attachment a log file from the yarplogger - it's pretty big (unzipped 80MB!) and it was just one minute  of movement!
[log-error-direct-pos-control.log.tar.gz](https://github.com/robotology/icub-firmware/files/388967/log-error-direct-pos-control.log.tar.gz)

Is it normal to get so many errors?
How can we fix the problem of the wrist not moving?

Thanks in advance

valegagge, 2016-07-29:
I Serena,
reading your log file, I can suppose not all boards have received the command of switch in position direct control mode due to high rate of commands sending.

In oreder to check if my conjecture is right, in your application , can you give setControlMode command to one joint per time and insert a small delay between two consecutive commands?
After that, you can check that every joint is in required control mode, before start to send setpoits.

Yu can perform these operations using your application or yarpmotorgui as you prefer.

Ciao
Valentina

barbalberto, 2016-08-01:
The following error message:
`number of streaming intput messages to be read is`  **7415**  `and can overflow `
is coming from the controlBoardWrapper receiving too many messages from its port so the queue is growing too fast with respect to the rate it is able to forward the commands to the HW. Closing the application which sends the data, the queue will be consumed and the joints will eventually move.

Also the other messages:
from eoMotionControl:
`HostTransceiver::addSetMessage for BOARD /w IP 10.0.1.2 successful ONLY at attempt num  2 with id:  ID32 = 0x01000b12 -> IND = 11, TAG = eoprot_tag_mc_joint_cmmnds_setpoint`

and from EMS:
SYS: the TX phase of the control loop has last more than wanted.

suggests to me that too many msgs are flooding the network.

When this errors appear, check the frequency the commands / rpc are sent or if there may be something slowing down the execution of commands.

marcoaccame, 2016-08-17:
To further support what Valentina suggests (send the control modes one at a time, add a small delay between each command,  read back the control mode before you go forward...) I say the following.

From the log I can see that the message `the TX phase of the control loop lasts more than wanted`, appears for boards EB2 / EB4 only and is due to the extra time taken to transmit 12 CAN frames on CAN1. 
These 12 frames are the commands which ask the 12 joints to go in direct position control mode. 
Each of the 3 MC4 boards thus receives a burst of 4 of such commands and I see that the boards fail to process the commands. 
In redmine issue https://redmine.icub.iit.local/issues/130, we already understood that the MC4 are unable to process burst of messages, probably because of the limited size of the RX CAN buffer. 

Hence,  Valentina's suggestion is due to both good practice (send a command and verify its effect) and to overcome a possible bug in MC4 CAN boards. 

Moreover, to make things work it is also important what Alberto suggests: try lowering the rate of commands / rpc of your application to avoid overflow at the controlBoardWrapper level.

 
Finally, if you don't have any more problems, I would close the issue in a few days.

closed 2016-08-29 by marcoaccame