robotology/icub-firmware #76: Signal when joint reaches joint limit

francesco-romano, 2016-09-12:
I would like to propose a feature request.

It can be very useful to have a log (error?) when the robot hits some joint limits.
Consider for example when the robot is in torque mode. 
The firmware does not apply the desired torque when the limit is hit, but because no errors is emitted understanding the cause of the robot misbehaviour can be very difficult (also because usually torque-based controllers are quite complicated..)

cc @randaz81 @marcoaccame 
@DanielePucci @traversaro

randaz81, 2016-09-12:
It can be done.
Some additional considerations:
- why you are interested in this? Actually the same problem applies to position/velocity control, but no signals are emitted, the trajectory is simply clamped.
- In torque mode the risk is to generate this signal several tens of times in a second.
- An high level controller can easily handle this: just check the encoder position vs the joint limits returned by the iLimit yarp interface (+ additional threshold)

valegagge, 2025-08-14:
@ale-git: If I'm not wrong, this feature is already implemented. Is it correct?