robotology/icub-firmware #83: The head and some arms joints cannot be restarted when pushing the fault button

S-Dafarra, 2018-02-08:
When pushing the fault on ``iCubGenova02`` or ``iCubGenova04`` some joints cannot run until a full reboot. In particular this happens all the time for the head joints and sometimes for the arms joints starting from the ``wrist_pronosup``.
Here a log 
[log_icub-head_yarprobotinterface_10797.txt](https://github.com/robotology/icub-firmware/files/1706773/log_icub-head_yarprobotinterface_10797.txt)

S-Dafarra, 2018-02-08:
cc @gabrielenava @marcoaccame @valegagge

marcoaccame, 2018-02-14:
Hi @S-Dafarra, thanks for the notification. We looked at the problem straight away last week and we have already tested a solution on the green robot. 

The source code is already committed (https://github.com/robotology/icub-firmware/commit/f3c3b94470eceecffbdc9b4e81b93e1222837e39) and the binary will be committed within the week. We are in the process of merging devel in master and hence, we prefer to pay attention to a new binary release.

I will alert you when everything is ready.

S-Dafarra, 2018-02-14:
Thanks a lot!

marcoaccame, 2018-02-14:
tested on green icub. problem solved. required fw version of mc4plus >= v3.14. it already in icub-firmware-build release branch. 

closing issue.

closed 2018-02-14 by marcoaccame