robotology/icub-firmware #87: Increase the PWM frequency in order to reduce the current ripple due to PWM modulation

nunoguedelha, 2019-07-15:
We suggest here an improvement for reducing the motor current ripples due to the PWM duty-cycle modulation.
The PWM ON/OFF phases result in a current oscillation which can reach 100mA amplitude, as initially analyzed and described in [this issue](https://github.com/dic-iit/element_torque-control-via-current/issues/61):

> The 100mA amplitude noise is actually the normal oscillation due to the PWM voltage excitation filtered by the phase inductance (~0.200mH). The same results can be obtained in a simulation where we feed with a PWM a R-L circuit having the same R, L values specified in the motor MOOG C2900584 datasheet as an example. Actually, as per prior tests run by @maggia80 's team, a single phase would exhibit an oscillation of ~1A peak-to-peak (I have a copy of the tests but not a link). But in a 3 phases Star configuration, the combination of the 3 respective PWM sources results in shorter PWM "ON" cycles, and so, smaller current ripples.

- The current ripples can be reduced by increasing the PWM frequency (above 20 kHz), regardless of the frequency of the Field Oriented Control task running on the FOC (20 kHz)
- This should not affect the settling time of the current as shown in the figure below (refer to section 2.3.3 and Fig.2.16 of my PhD dissertation):

## Example of current response simulation on a single phase (R = 0.341 Ω, L = 0.225 mH)

### PWM freq = 20 kHz
[image]
Current ripple amplitude = 2 A

### PWM freq = 80 kHz
[image]
Current ripple amplitude = 0.5 A

nunoguedelha, 2019-07-22:
In order to proceed with the implementation we need to verify the following points:
- [ ] have we tried before to increase the PWM and/or the DMA interrupt task frequency to 40kHz?
- [ ] what is the maximum frequency available?
- [ ] where is this properly documented (datasheet, section,...)?
- [ ] I could find the datasheet of the dsPIC33FJ128MCX02 in the Microchip site [here](http://ww1.microchip.com/downloads/en/devicedoc/70291g.pdf). That datasheet does not explain clearly how to configure the PWM frequency. Do we have other documentation on this?

@ale-git , we need to answer these questions ASAP in order to proceed with the implementation and detect problems soon enough to allocate more time to this task if it requires extra effort.

nunoguedelha, 2019-07-26:
@ale-git showed which parameter to set in the 2FOC code for changing the PWM frequency. I'm documenting here how it works and relates to the Motor Control PWM (MCPWM) settings described in  the manual http://ww1.microchip.com/downloads/en/DeviceDoc/70187E.pdf.

The MCPWM1 module is the one used for driving the motor. Its time base interrupts and output modes are configured through the registers PxTCON, PxTPER, PWMxCON1, PxDTCON1, PxDC1, PxDC2, PxDC3 by the function call `pwmInit(LOOPINTCY/2, DDEADTIME, (8*LOOPINTCY)/20 /*pwm max = 80%*/);`.

• PxTCON: PWM Time Base Control Register
This register is used for the selection of Time Base mode, time base input clock prescaler,
time base output postscaler, and for enabling the time base timer.
```
PTCKPS<1:0> - PWM Time Base Input Clock Prescale Select bits: 00 = PWM time base input clock period is TCY (1:1 prescale)
PTMOD<1:0> - PWM Time Base Mode Select bits: 10 = PWM time base operates in Continuous Up/Down Count mode
```

• PxTMR: PWM Time Base Register
The time base count value and the time base count direction status are obtained in this
register.

• PxTPER: PWM Time Base Period Register
The PWM time base value is written into this register, which determines the PWM operating
frequency.
```
PTPER<14:0> - PWM Time Base Period Loop Value bits: LOOPINTCY/2
```
Note: The PWM period setting register is set to LoopInTcy/2 but since it counts up and
then down, and the interrupt flag is set to 1 at zero, the actual interrupt period is `LOOPINTCY`.

• PWMxCON1: PWM Control Register 1
Selection of either Independent or Complementary mode for each PWM I/O pair is
performed in this register.
```
= 0x0077 (Enable PWM 1,2,3 pairs (L+H) for complementary mode)
PMOD4:PMOD1 - PWM I/O Pair Mode bits: 0x00 = all enabled PWM I/O pin pair is in Complementary Output mode
PEN4H:PEN1L - PWMxHy I/O Enable bits: 0x77 = Enable PWM 1,2,3 pairs
```

• PxSECMP: Special Event Compare Register
This register provides the compare value at which the analog-to-digital conversions are to
be synchronized with the PWM time base. Comparison can be either during up-count or
down-count in Center-aligned mode depending on the setting of the SEVTDIR bit in this
register.
```
(Phase of ADC capture set relative to PWM cycle: when arrive to PTPER offset and counting down to avoid ripple on the current measurement)
SEVTDIR - Special Event Trigger Time Base Direction bit: 1 = Special Event Trigger will occur when the PWM time base is counting down
SEVTCMP <14:0> - Special Event Compare Value bits: 0x00 = at the top of the current ripple
```

• PxDTCON2: Dead Time Control Register 2
Dead time insertions from Dead Time Unit A or Dead Time Unit B for each of the PWM
outputs can be selected using this register.
```
DTS4A:DTS1I - Dead Time Select bits for PWM Signal Going Active: 0 = Dead time provided from Dead Time Unit A
```

• PxDTCON1: Dead Time Control Register 1
The dead time value and clock period prescaler for Dead Time Unit A and Dead Time Unit
B can be selected using this register.
```
= (0x00 | (PWM_DEADTIME)); // Dead time. Prescaler 0
DTAPS<1:0> - Dead Time Unit A Prescale Select bits: 00 = Clock period for Dead Time Unit A is TCY
DTA<5:0> - Unsigned 6-bit Dead Time Value bits for Dead Time Unit A:  DDEADTIME
```

• PxFLTACON: Fault A Control Register
This register provides the following selections:
- PWM output pin driven on an external fault – active or inactive state
- Fault mode – Cycle-by-Cycle mode or Latched mode
- Pin pair to be controlled or not controlled by Fault Input A
```
= 0x00 (PWM fault pins not used)
FAEN4:FAEN1 - Fault Input A Enable bit: 0 = PWMxHy/PWMxLy pin pair is not controlled by Fault Input A
```
[... 96 lines cut ...]
```
PWMq_ems = Vq_yarp / 100 x PWM_50_DUTY_CYCLE
```
`Vq_yarp` ranging from negative to positive values.

This value is initially encoded in 16 bits (0xFFFF = d32767), so the closest we can get from the max value is by shifting 5 bits left:
```
CtrlReferences.VqRef = PWMq_ems << 5 = Vq_yarp / 100 x PWM_50_DUTY_CYCLE x 2^5
```
... hardcoded as
```
CtrlReferences.VqRef = Vq_yarp / 100 x 32000
```
converted to...
```
VqRef = ((long)CtrlReferences.VqRef)<<(10-VOLT_REF_SHIFT) = ((long)CtrlReferences.VqRef<<10)>>5
```
before use it in the PID computations. And back to:
```
Vq = VqRef>>10
```
So the overall conversion is as follows:
```
Vq = Vq_yarp / 100 x PWM_50_DUTY_CYCLE
```

nunoguedelha, 2019-07-26:
## Setting the new PWM frequency

The new PWM frequency is set through the flag defined in `UserParms.h`:
```
#define PWMFREQUENCY  40000
```

## Issues and Fixes

- The function call `pwmInit(LOOPINTCY/2, DDEADTIME, (8*LOOPINTCY)/20 /*pwm max = 80%*/);` sets `PxTPER = LOOPINTCY/2`, but should set `PxTPER = LOOPINTCY/2-1`.
This issue could cause eventually a PWM offset.
- The half fullscale of the PWM
```
CtrlReferences.VqRef = (Vq_yarp / 100 x PWM_50_DUTY_CYCLE)<<5
```
...  is hard-coded as follows in the EMS controller:
```
CtrlReferences.VqRef = Vq_yarp / 100 x 32000
```
but should instead, depend on PWM_50_DUTY_CYCLE. So we adjust the computation as follows:
```
Vq = Vq*PWM_50_DUTY_CYCLE/1000;
Vd = Vd*PWM_50_DUTY_CYCLE/1000;
```

nunoguedelha, 2019-07-26:
Updated the expression of `Vq` and `Vd` since they are integers, and...

- for `Fpwm = 20 kHz`, `1000/PWM_50_DUTY_CYCLE = 1`
- for `Fpwm = 40 kHz`, `1000/PWM_50_DUTY_CYCLE = 2`

this way I'm sure not to have overflow or round issues.

nunoguedelha, 2019-07-26:
The proper way to do it, also from a performance point of view, would be to do it in the CAN parser `s_canIcubProtoParser_parse_pollingMsg` where `CtrlReferences.VqRef` is set. But this would impact the behaviour of the closed loop since it would change the ratio between the PWM delta (Kp*IqError) and the PWM open loop values.

The feed-forward component implementation in the control loop will avoid that problem. So we can implement and release the proper PWM re-scaling along  with the feed-forward.

CC @ale-git

nunoguedelha, 2019-08-02:
# Results

### Probed PWM voltage in one of the terminals (terminal 1 as done [here](https://github.com/dic-iit/element_torque-control-via-current/issues/62#issuecomment-494382216))
[image]
As we can see in the above diagram, the PWM period is $25 \\, \mu s \sim 40 kHz$ (channel 3).

### Received  and Transmitted CAN messages
We don't see any impact on the exchange of CAN messages. We used a counter to check for lost or delayed packets.

### ms counter in the transmitted CAN messages from the 2FOC (20 kHz)
[image]

### ms counter in the transmitted CAN messages from the 2FOC (40 kHz)
[image]

The counter is incremented in the `CanIcubProtoTrasmitterSendPeriodicData()` where the CAN message is built. As expected the CAN message is sent every 1ms. The observed delays occurrences don't get worse with the PWM frequency increase.

The interesting would be to measure the minimum round-trip delay between the downlink parsed `CtrlReferences.IqRef` and the uplink `I2Tdata.IQMeasured` in the case where we assume a perfect controller just setting `I2Tdata.IQMeasured = CtrlReferences.IqRef`.

nunoguedelha, 2019-08-07:
Implemented in https://github.com/robotology/icub-firmware/pull/90.

marcoaccame, 2019-08-08:
the feature is in devel now

closed 2019-08-08 by marcoaccame