robotology/icub-firmware #89: Analyze residual noise at zero phase current

nunoguedelha, 2019-08-05:
When we set off all the PWM input signals, i.e. all terminal to terminal voltages are null, we should measure a null current, but instead we get a noisy measurement with a peak to peak amplitude of ~160mA and a quantization of ~20mA.
We need to check if this resolution and noise level is normal with respect to the expected performance of the current sensing module: hardware characteristics and configuration of the hall effect sensor in series with the ADC module.

nunoguedelha, 2019-08-05:
The experiment was done using an `icub-firmware` debug build logging the phase currents `qIa` or `qIc`.

The debug build can be found in the [`icub-firmware-build` repo, branch `issue/#89_phase-current-noise`, file `CAN/2foc/2foc.hex`](https://github.com/robotology/icub-firmware-build/blob/issue/%2389_phase-current-noise/CAN/2foc/2foc.hex).

The used branch can be found in the fork repo `https://github.com/nunoguedelha/icub-firmware.git`, debug branch `debug/log-phase-currents`, commit https://github.com/nunoguedelha/icub-firmware/commit/d9342283706daa842883d6ca0bb10cee792b0569.

The results are similar for both phases `qIa` and `qIc` (anyway, regarding `qIb`, in the 2FOC controller, `qIb=-(qIa+qIc)`.
[image]

The measurements are in `mA`. We can see the quantization at 20mA.

Next step is to check the Hall sensor and ADC configurations.

nunoguedelha, 2019-08-05:
# Check the Hall Effect current sensor connection and datasheet

- 2FOC logic schematics: https://svn.robotology.eu/repos/iCubHardware/trunk/electronics/private/P2009_05_2FOC/Documentation/Presentations/2FOC%20-%20EN.pdf
### ACS714 connection (page 6)
[image]
(This can be helpful if we wish to check the output signals OUPUT(7)-IU1 and OUPUT(7)-IW1 waveforms.)

The used Hall effect sensor is the ACS714. The FILT(6) output is set to 47nF.
 
- The [ACS0714 datasheet](https://svn.robotology.eu/repos/iCubHardware/trunk/electronics/private/P2009_05_2FOC/Datasheets/CurrentSense/ACS0714.pdf) specifies a sensitivity of 100mV/A (page 6, x20A PERFORMANCE CHARACTERISTICS) for an Ip = 20A (Optimized Accuracy Range).
As we saw in the 2FOC logic schematics (FILT(6) -> 47nF), the bandwidth is set to 2kHz, for which the Vnoise(PP) = 11mV.

@ale-git How can we check which range is selected?

nunoguedelha, 2019-08-05:
# Check ADC module configuration

- The ADC module configuration can be found in the [ADC manual, section 16 of the 2FOC manual](https://svn.robotology.eu/repos/iCubHardware/trunk/electronics/private/P2009_05_2FOC/Datasheets/dsPIC33/Section%2016.%2010-12-bit%20ADC%20with%20DMA%20-%20dsPIC33F%20FRM.pdf), Register 16-2: ADxCON2: ADCx Control Register 2.

In the 2FOC controller code ( `ADCConfigPWMandDMAMode(...)` in `ADC.c` ), the ADC is configured in 10bits mode (`AD1CON1bits.AD12B = 0`), and VCFG<2:0> is set to 0 (refer to Register 16-1: ADxCON1). This implies that the resolution in Volts is defined by (AVDD-AVSS)/(2^10).

We have to check the values of AVDD and AVSS. @ale-git can you take this action?

### Register 16-1: ADxCON1: ADCx Control Register 1
[image]

### Register 16-2: ADxCON2: ADCx Control Register 2
[image]

nunoguedelha, 2019-08-05:
CC @marcoaccame @DanielePucci

claudiolorini, 2019-08-05:
As far as i can tell, the noise level you experiencing are 'normal' and intrinsic in the technology of the sensor (hes). This kind of noise (as far as it ha a 0 mean value) does not have relevant effect on the torque output but only audible effect.
I'd rather be more concerned of the non zero mean value that is generated by the ACS714-20 when there is a temperature change. This effect cannot be removed, but only mitigated calibrating the offset of the conversion chain (hes&adc) periodically. This kind of drift effects the torque ripple of the system.

nunoguedelha, 2019-08-08:
Thanks @claudiolorini for the details.
### Regarding the intrinsic noise...
indeed, as a side note, the [ACS0714 datasheet](https://svn.robotology.eu/repos/iCubHardware/trunk/electronics/private/P2009_05_2FOC/Datasheets/CurrentSense/ACS0714.pdf)(page 6) mentions a sensitivity of 100mV/A and peak-to-peak noise of 11mV for an Optimized Accuracy Range Ip = 20A, which implies a peak-to-peak noise of 100mA.

- But isn't this a high frequency noise that should be filtered by the FILT(6) -> 47nF ?

- Has this noise spectrum ever been characterized? I understand that there is no direct impact on the torque, but couldn't it impact the Current tracking if that tracking is done at 20kHz or 40kHz?

### non zero mean value...
@ale-git worked on the improvement of the HES+ADC chain offset and gains calibration but it is done only at startup. So as you advised, we need to implement a periodic calibration or a dynamic compensation that requires the identification of `offset = f(temperature)`.
- @ale-git has this drifting offset ever been addressed previously?

ale-git, 2019-08-08:
@nunoguedelha Giorgio Zini explained to me that with a sensitivity of 100 mV/A, reduced to 65.8 mV/A by the voltge divider made by R29, R30 and R103 resistors, and 3.3 V power supply, 10 bit converter, the LSB of the converter corresponds to 3.3 V / 1024 = 3.22 mV. Dividing by the sensitivity we obtain the current quantization that corresponds to the LSB: 3.22 mV / 65.8 mV/A = 50 mA of intrinsic error related to the LSB noise. The noise in the plot is reduced by the two subsequent sample filtering.
About the temperature drift, we could repeat the calibration any time the PWM is switched off, and every 20 seconds in OFF state.

nunoguedelha, 2019-08-21:
https://github.com/robotology/icub-firmware/issues/89#issuecomment-519438808
### Regarding the intrinsic noise...

> The noise in the plot is reduced by the two subsequent sample filtering.

Ok @ale-git , but I think we shouldn't see such noise in the first place with the filtering of C28 and C29 capacitors.

> * But isn't this a high frequency noise that should be filtered by the FILT(6) -> 47nF ?
> 

Ciao @claudiolorini , the capacitors C28 and C29 are actually missing on the test board I'm using. Are we sure they are present in the 2FOC boards on the robot?

### non zero mean value...

@claudiolorini @ale-git , is the module **"Dynamic Offset Cancellation"** active in the ACS0714 HES [(page 3 of the datasheet)](https://svn.robotology.eu/repos/iCubHardware/trunk/electronics/private/P2009_05_2FOC/Datasheets/CurrentSense/ACS0714.pdf)? Isn't it enough?

[image]

claudiolorini, 2019-08-21:
Boards in revision C does have the caps mounted:
"3336 . C - 2FOC, IIT - Electronic board with CAN interface, dual driver for brushless motors, power supply 18-48V, 20A"

Boards in revision F does not have the capacitors mounted:
"3336 . F - 2FOC, IIT - Electronic board with CAN interface, dual driver for brushless motors, power supply 18-48V, 20A full Band width, for iCub, new heatsink"
I suppose your robot uses this one.

The capacitors introduce a delay in the current control loop (20KHz), and this is definetly *BAD*.

By experimental tests, the internal offset correction is not 100% effective to avoid temperature or 
stray-fields related offets. 
Try heating the board or placig it near an iron part, you will see the current offset change.

nunoguedelha, 2019-08-21:
> Boards in revision F does not have the capacitors mounted:
> "3336 . F - 2FOC, IIT - Electronic board with CAN interface, dual driver for brushless motors, power supply 18-48V, 20A full Band width, for iCub, new heatsink"
> I suppose your robot uses this one.

Ciao @fabrizio69, can you please confirm that on iCubGenova04 (Greeny) we are using the 2FOC revision F?

marcoaccame, 2019-08-27:
Hi @nunoguedelha, I have just checked w/ @davidetome  & @miche19:
iCubGenova04 and most recent iCubs use **revision F**.

nunoguedelha, 2019-08-27:
Thanks @marcoaccame ! Sorry I hadn't update the issue, @fabrizio69 had also confirmed this.

nunoguedelha, 2019-08-28:
> @ale-git How can we check which range is selected?

As explaned by Giorgio Zini, the Ip current range depends on the model of the ACS714 IC, The one mounted on the 2FOC boards is the `ACS714LLCTR-20A-T`. A summary of the sensitivities depending on the models is given in the manual:
[image]
This confirms the 100mV/A sensitivity.

claudiolorini, 2019-08-28:
LLCTR-20A-T