robotology/icub-firmware #96: Define, implement and test a calibration procedure for setting the required PWM offsets

nunoguedelha, 2019-08-30:
After analysis in https://github.com/dic-iit/element_torque-control-via-current/issues/59, we concluded that the motor quadratic current oscillations depending on the rotor position were due to the current offsets measured on the stator terminals (phases), while the digital PWM values computed by the inverse Clark-Park transform have no offsets, and are phase shifted by 120 degrees as expected. The offsets could be measured also on the oscilloscope. As the HES/ADC offsets had already been compensated, the current measurements offsets are solely due to the duty-cycle offsets in the PWM voltages generated by the MOSFET Bridge transistors driving the three phases of the stator.

We define and design here a procedure strategy for identifying and compensating those offsets:
- at the startup phase of the 2FOC driver,
- after the HES/ADC output offsets have been compensated.

## Assumed pre-compensation of the HES/ADC output offsets
We consider that the HES/ADC offsets, mostly due to magnetic hysteresis in the HES core, and eventually by a not perfectly centered ADC output with respect to the full range of voltage inputs coming from the HES output (ex: IU1-RA0, IW1-RA1).

So we assume here that for a 0 A real current going through the HES sensor (pins PX1, PZ1), i.e. PWM is set to off on all phases, the current measurement on terminal A or C will return 0 A (a call to `MeasAndCompIaIcCalculateIb()` will set `ParkParm.qIa` and `ParkParm.qIc` to 0).

## Compensation of the PWM offsets
It is important to distinguish the HES/ADC offsets from the PWM offsets for the following reasons:
- The duty-cycle offsets in the PWM voltages generated by the MOSFET Bridge transistors result in real current offsets that cannot be compensated by a calibration of the HES/ADC measurement chain.
- We expect the HES/ADC offsets to drift over the sensor temperature, while the PWM offsets should be stable. If the later also drift, they would probably have a drift rate different from the the HES/ADC drift, requiring different compensation solution.

### Proposed compensation approach
...

nunoguedelha, 2019-09-02:
@ale-git @marcoaccame

ale-git, 2019-09-03:
@nunoguedelha I tried to compensate both A and C PWMs with a single calibration, but it didn't work, probably because of the mutual influence. So I think that the separate calibration that you propose is more promising. It is easy to separate the double calibration of the test firmware in two single calibrations.

ale-git, 2019-09-03:
@nunoguedelha let's discuss it in the afternoon as planned.

nunoguedelha, 2019-09-03:
Thanks @ale-git . I won't be able to come to IIT, but we can chat on skype or telegram as you prefer.

ale-git, 2019-09-06:
@nunoguedelha @marcoaccame 

I implemented a PWM calibration procedure (step 2 in the following calibrations list):

step 1) current calibration (as usual): with PWM off the current on phases A and C is measured as the average of 1000 readings and the values are used as current offset.

step 2) with C phase "disconnected" (transistors off) and B phase 50%, A is adjusted until Ia = 0 current is measured. Then phase A is disconnected and phase C is adjusted until Ic = 0 current is measured.

step 3) phase B is disconnected and phase A is given a PWM increment above 50%, and Ic is given a PWM decrement under 50% to generate a current. With B disconnetced Ia must be physically equal to -Ic, so the current sensing gains Ka and Kc are adjusted so that Ia = -Ic.

Step 1 and 3 are working fine, but step 2) gives different offset results for each calibration. The following plots depict the effect of the PWM calibration: a further current oscillation on a electrical cycle. The interesting thing, worthy of further study, is that the noise amplitude is well position dependent, as if the noise on phase A and C sums up for certain positions and cancels for others.

[image]

[image]