robotology/icub-firmware #102: bug in driver can in can::transmit

valegagge, 2019-10-31:
While I was managing this [issue](https://github.com/robotology/icub-tech-support/issues/854) , I understood that **the driver CAN on strain2 has a bug**. I noticed that in `embot :: hw :: can :: transmit` IRQ protection is missing: both task and CAN IRQ handler can access to transmission queue at same time.

Therefore, it is important to update such function in order to avoid the race condition.
Since the `hardware abstraction layer` is common between more boards in important to fix it.

@marcoaccame told me that he saw a crash in `embot :: hw :: can :: transmit` during RFE application debugging.

The CAN driver has been developed presuming that first the strain application collects all data, then fills in one or more frames and then transmit all together. This behavior is similar to EMS board's application.  
Actually, the strain application can't separate the phase of data collecting from transmission one, so I think that the CAN driver should be updated in order to provide different sets of functions, ie different possible behaviors, to allow the application to use the most suitable for itself.

***I'd like to point out that we need to fix the bug in first step, then we could improve the driver making it multiple purpose*** as explained above.

DanielePucci, 2019-11-13:
Hi @valegagge by chance any update on this?

valegagge, 2019-11-18:
Hi @DanielePucci , 
I notified @marcoaccame about this fix.

I already studied one possible solution and asap @marcoaccame and I will work on this issue.

marcoaccame, 2019-11-22:
Hi, here are some updates.

## step 1: define the job
- we are going to use a branch in robotology/icub-firmware with the fix to the can driver
- i will build a test to force a situation which makes a CAN board crash so that we can verify the fix

**DoD**
- by launching the test the board crashes before the fix  and it does not crash after.
- do a proper fw release of all CAN boards

## step 2: the job

### find a test which makes the board crash

I cannot recreate the crash at my desk.

here is what happened

```
I have done some tests, but the problem does not show.
I have hence decided to run the strain2 board in the same mode as in iCubGenova04 `
and ... see what happens after some hours. I have launched at 1430 today. 
let's see tomorrow morning
```

```
Ok, it is about 18 hours after I launched the board.
here is what i found:

the strain2 board is blinking.
i attach the can-usb but i dont see any transmission from strain
i have tried to send a discovery message and the board replies
ok, something has happened.

the board has not crashed because it replies to discovery and its led blinks.
however it does not TX anymore.
It is it possible that its power went off during the night and it restarted losing 
its configuration.

```

what I will do next is:

repeat once more with exactly the same application version as in iCubGenova04
add a timeoflife variable which i can quesry so that i can compute the time of last bootstrap and then redo the experiment.

### apply the fix to teh test to verify that it is effective

TBD

marcoaccame, 2019-11-26:
I have found a way to reproduce the bug systematically:

- i compile with -O0 optimization,
- i run a stress test with a .cspsl file which sends a great amount of CAN messages.

yes, the problem is a concurrent access to the tx fifo from the user thread and the IRQ Handler.
I have already tested a solution, which i will soon push into robotology/icub-firmware, devel branch.

S-Dafarra, 2019-11-26:
That's great!

marcoaccame, 2019-11-26:
Hi @S-Dafarra, i will give you a binary to test on the robot by today

S-Dafarra, 2019-11-26:
Currently, we are using a firmware version with some developments by @nunoguedelha. One of the things that changed is that all the gains are set to positive. Thus, before testing, we need also to change the ``robots-configuration`` branch.

marcoaccame, 2019-11-26:
hi @S-Dafarra, you just need to update the FW of the `strain2` according to https://github.com/robotology/icub-firmware-build/commit/335ddc0e47b723c278905f84b30d8d1e822ea756

the developments done by @nunoguedelha affects only the `ems` and `mc4plus` boards and can safely coexist with the new `strain2` version.

S-Dafarra, 2019-11-26:
Right, I forgot. Thanks!

S-Dafarra, 2019-11-28:
Today I flashed the FW. I just ran the robot once to check that everything was ok and it started with no errors. We will exploit future tests on the robot to check if the problem have been definitively solved. Thank you.

marcoaccame, 2019-11-29:
Closing issue.

closed 2019-11-29 by marcoaccame