robotology/icub-firmware #104: Avoid glitch while switching control mode

pattacini, 2019-12-05:
From https://github.com/icub-tech-iit/task-force-miscellanea/issues/6#issuecomment-561711268

>When the control mode is changed for a joint the PID is reset (i.e. its integral part is reset), so there is a little glitch until the integral part is accumulated again. It is safe, but there is a visual effect. Probably, we could avoid resetting the PID but it must be tested carefully.

cc @traversaro @xEnVrE @marcoaccame

traversaro, 2019-12-05:
cc @GiulioRomualdi @S-Dafarra

traversaro, 2019-12-05:
According to @S-Dafarra , this could explain the oscillations at second 34 of this video: https://www.youtube.com/watch?v=33F2G5mYiHY&feature=youtu.be&t=33 .