robotology/icub-firmware #105: 2FOC quadrature encoder error

valegagge, 2020-02-07:
On icubV3 I experienced of following error when running `yarprobotinterface:

```
DEBUG]yarprobotinterface running happily
[ERROR] from BOARD 10.0.1.4 (right_arm-eb4-j2_3), src LOCAL, adr 0, time 164s 766m 779u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .

[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 166s 448m 665u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please ch
```

when I relauched `yarprobotinterface` I get following errors:
```
Poi ho spento i motori e ho rilanciato yarprobot interface e avevo questi errori:
[ERROR] from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 217s 849m 768u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet

[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 219s 125m 408u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[ERROR] from BOARD 10.0.1.2 (left_arm-eb2-j2_3), src LOCAL, adr 0, time 219s 302m 419u: (code 0x0200000d, par16 0x0001 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[ERROR] from BOARD 10.0.1.3 (right_arm-eb3-j0_1), src LOCAL, adr 0, time 220s 946m 908u: (code 0x0200000d, par16 0x0000 par64 0x0000000000000000) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
[ERROR] from BOARD 10.0.1.4 (right_arm-eb4-j2_3), src LOCAL, adr 0, time 221s 623m 599u: (code 0x02000008, par16 0x0000 par64 0x0000000000000002) -> MC: AEA encoder invalid data. Hardware problem in the magnetic position sensor of the joint caused invalid position readings. par16 = AEA port (msb) and ID of joint (lsb). + .
```

As the logs reported, the encoder in error changes almost every run of `yarprobotinterface`.

I already had a chat f2f with @ale-git, who suspect a wrong configuration of encoder filters.

I started to notice this error when I update the last firmware version.
Here there are the versions of firmware:
[image]

[image]

vtikha, 2020-03-02:
Is it possible to have an update or look at this issue as we cannot use iCub3 and therefore cannot go on with our work. 

cc @ale-git

ale-git, 2020-03-02:
I'll take a look tomorrow.

ale-git, 2020-03-03:
The AEA sensor of joint 2 (yaw) of the right arm is not working for sure, probably the sensor must be replaced.

About the QE error I can't understand yet what's going on. 
@valegagge can you please detail the firmware updates at the origin of the problem? Did you update the 2FOC firmware?

valegagge, 2020-03-03:
@ale-git 
I'm not totally sure but the error on AEA is not more present,  but the QE problem continues occur.

We experienced of this error after I updated the all firmware on the robot, 2FOC included.
From the image I reported before you can see the version of new firmware of 2foc, but I'm sorry I haven't the previous version.

When we had a f2f chat, you told me that the issue could be about some filter of encoder QE readings.
Did you updated something about it?
Moreover this type of issue appear and disappear on different joints ant lets me think that encoders are not broken, but there is some "electrical issue" or some difference in encoder reading filtering.
So, if you didn't change nothing in the firmware about QE readings maybe it could be some hardware related issues.

ale-git, 2020-03-03:
Hi @valegagge the AEA error comes and goes, but it is present for the most part. The encoder error is present at calibration time but after you force idle all the joints work fine, it's not persistent. There must be a reason for it anyway, but yet I can't figure out why.

valegagge, 2020-03-09:
Hi @ale-git ,
in the last week I did some tests on robot and I noticed that the problem I reported happen when launch `yarprobotinterface` after switched on the motors. So if you follow this procedure you can experice of same errors:

1. switch on motors
2. launch yarprobotinterface 
3. you get one or more joint in error: `MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables ....`
4. put in `idle` and then in `run` using `yarpmotorgui`, now the joint are running correctly.
5. stop `yarprobotinterface`
6. run again `yarprobotinterface` without switch off and on the motors
7. now the joint taht before were in fault now are in position mode.

I repeated this procedure and I saw the same error on different joints, but the behavior is the same.

I hope that this result can halp you to detect the origin of this issue.

Moreover I want to tell you taht no error on AEA had been reported during these tests

valegagge, 2024-02-02:
Closing. No longer useful.

closed 2024-02-02 by valegagge