robotology/icub-firmware #114: Locate the coupling matrix between finger joints and motors values

kouroshD, 2020-04-08:
There is a time invariant linear mapping from the motor values to the joint values of the iCub fingers as following:
$$
S_{hand}= A M_{hand}
$$
Where $S_{hand}$ and $M_{hand}$ are the vector of joint values and motor values at each moment, and $A$ is the fixed mapping when there is not any contact.
My questions are:
- If somewhere in firmware we compute $A$ matrix?
- If somewhere $A$ matrix has been exposed?

GiulioRomualdi, 2020-04-08:
Hi @kouroshD I found [this coupling](https://github.com/robotology/robots-configuration/blob/be62f5e8268c6c90c027290bc09c735ca00286fa/iCubGenova04/hardware/mechanicals/left_arm-eb26-j12_15-mec.xml#L32-L51) in the configuration file.

kouroshD, 2020-04-09:
@GiulioRomualdi  Thank you for the links, they were interesting. However, as far as I see the values inside the matrices does not coincide with the description of mechanical coupling of the hand according to this paper: [A. Schmitz, U. Pattacini, et.al., "Design, Realization and Sensorization of the Dexterous iCub Hand"](https://ieeexplore.ieee.org/abstract/document/5686825)