robotology/icub-firmware #121: Import FW sources for BDC BCB boards

davidetome, 2020-08-07:
## Task description
Import/migrate the firmware sources for ``bdc`` and ``bcb`` boards from : 

- https://github.com/icub-tech-iit/electronics/tree/master/projects/boards/P2013_09_BDC/sw/BDC
- https://github.com/icub-tech-iit/electronics/tree/master/projects/boards/P2013_03_BCB/sw/BCB

## Definition of Done
the migraation is done

cc @marcoaccame

davidetome, 2020-08-20:
see PR https://github.com/robotology/icub-firmware/pull/125

closed 2020-08-20 by davidetome

MrAndrea, 2020-08-21:
Ciao @davidetome,
there are 2 firmware for the 2 revision of the BAT board, since they have different microcontroller.
For the first and obsolete revision, the firmware is in:
https://github.com/icub-tech-iit/electronics/tree/master/projects/boards/P2015_06_BAT/sw/BAT

For the latest board revision (iitcode 10050.C), the firmware is in:
https://github.com/icub-tech-iit/electronics/tree/master/projects/boards/P2015_06_BAT/sw/BAT%20Rev%20B

davidetome, 2020-08-25:
@MrAndrea I moved also the `BAT Rev B` folder, please take a look at the [PR](https://github.com/icub-tech-iit/electronics/pull/45) and use `squash and merge`  instead `merge` as pointed out by @pattacini

pattacini, 2020-08-25:
https://cloudfour.com/thinks/squashing-your-pull-requests/