robotology/icub-firmware #152: Fix building problem - MPLAB

giovannipizzolante, 2020-12-15:
When building the 2FOC.V3 project, the following error appears:

```../src/ecan.c:827:37: error: conflicting types for 'int8_t'```

 It is a problem related to the new compiler version. The old version contains this line:

https://github.com/robotology/icub-firmware/blob/b6c0a716bb75105fd1d62cfce6ae05c7d649df9b/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/ecan.c#L827

that causes that error.

The fix is:

https://github.com/isorrentino/icub-firmware/blob/afd340abb343c5d5502970961266feaac327dd8a/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/ecan.c#L1135

GiulioRomualdi, 2020-12-15:
cc @marcoaccame @traversaro @isorrentino @nunoguedelha

traversaro, 2020-12-15:
Which version of MPLAB are you using? Did you use the projects in the icub-firmware directly or you modified them in someway?
 
The problem seems due to the fact that there are two definitions of `int8_t`, and probably due to a newer version of the compiler/standard library the definition of `int8_t` changed from `char` to `signed char`, but the problem is that that typedef should not be there. Since C99, `int8_t` is defined in `stdint.h`, so it should be just sufficient to include that header.

isorrentino, 2020-12-15:
>  Did you use the projects in the icub-firmware directly or you modified them in someway?

I just cloned the repo, imported the project in MPLAB and tried to compile it.

> Which version of MPLAB are you using?

- Product Version: MPLAB X IDE v5.40
- Compiler version: MPLAB XC16 Compiler v1.50

traversaro, 2020-12-15:
Probably it make sense for reproducibility to mention somewhere in this repo or in https://github.com/robotology/icub-firmware-build which version of the MPLAB IDE/Compiler is used to generate the firmware. cc @marcoaccame @Nicogene @davidetome @ale-git

marcoaccame, 2020-12-18:
the correct place where to put the compiler version used to generate the binary is in the log of the binary itself.
that already contains some basic info about how to reproduce the same binary, see [here](https://github.com/robotology/icub-firmware-build/commit/46d3f477b566be0b01d34409987e5d259abe36bb), and we could also add the version of the toolchain.

isorrentino, 2021-01-28:
The PR has been merged. Closing.

giovannipizzolante, 2021-01-28:
Sure @isorrentino . Closing the issue

PR #158 (2021-01-28): 2FOC Firmware -  fix compilation error
This fix #152 . As suggested by @marcoaccame the compilation error has been fixed by removing ```typedef``` lines and putting an ```#include "stdint.h"``` .
closed 2021-01-28 by giovannipizzolante