robotology/icub-firmware #160: Add the possibility pass the root of robots-configuration in FirmwareUpdater.script.info.all.sh

GiulioRomualdi, 2021-01-28:
`FirmwareUpdater.script.info.all.sh` searches for `robots-configuration` folder inside `$ROBOT_CODE` i.e. 

```sh
./manageFWrobot.py -n $ROBOT_CODE/robots-configuration/$YARP_ROBOT_NAME/network.$YARP_ROBOT_NAME.xml -f ../info/firmware.info.xml -p all -a info | tee ../logs/log.of.FirmwareUpdater.$YARP_ROBOT_NAME.info.all.txt
```

We usually install `robots-configuration` as part of the superbuild, so our path is `$ROBOTOLOGY_SUPERBUILD_DIR/src/robots-configuration`. It would be nice to pass the root of robots configuration as parameter to the `sh` script.

GiulioRomualdi, 2021-01-28:
I just noticed that I should open it in icub-firmware-build.

traversaro, 2021-01-28:
Note that 

> We usually install `robots-configuration` as part of the superbuild, so our path is `$ROBOTOLOGY_SUPERBUILD_DIR/src/robots-configuration`.

Note that since iCubWaterloo01/SN44, the iCubs that come out from the "production" also use the superbuild, so this is a problem that affects also those robots.

GiulioRomualdi, 2021-01-29:
If you tell me what you prefer I can open a PR. Should I close this issue and open a new one in icub-firmware-build?

traversaro, 2021-01-29:
> I just noticed that I should open it in icub-firmware-build.

I transferred the issue there.