robotology/icub-firmware #161: Document how to configure the environment to compile the EMS board firmware

isorrentino, 2021-01-30:
This issue follows from https://github.com/dic-iit/lab-organisation/issues/148#issuecomment-770248640. I copy here the comment as I do not know if everybody has access.

> On Thursday I needed to build the firmware of the EMS board and I had to ask @marcoaccame for help. Only now I discover the existence of @nunoguedelha's comment in this issue in which all the steps necessary to properly configure the ide and the project are documented. I think that in the future someone else may need this guide, so it would make sense to put it somewhere that is more easily accessible, such as in the `README` of `icub-firmware` as @nunoguedelha also proposed.

traversaro, 2021-02-01:
@isorrentino if there is already some material already available in that issue, I think the first step could be to copy it here so that anyone interested can clean it up and add it to the docs.

isorrentino, 2021-02-01:
## Comments from https://github.com/dic-iit/lab-organisation/issues/148

- ### Nuno's comments
We are installing the development environment for the EMS board (ARM) from `Keil`, on Windows on the Monster machine.
Online link to the app: https://www.keil.com/demo/eval/arm.htm
Setup file location: `C:\Users\icub\Documents\ProgramInstallFiles\mdk528a.exe`
Install folder: `C:\Keil_v5`

How-to for the license activation: https://www.keil.com/product/brochures/readmefirst.pdf

**Note:** It is strongly recommended to accept the target install folder proposed by the installer, otherwise the activation phase will likely fail with the error message "TOOL.INI not installed".

Thanks @marcoaccame !!

I'm listing here the additional steps done by @marcoaccame for the installation of the IDE:

**download the `icub-firmware` and `icub-firmware-shared` repositories**

... downloaded into `C:\dev\git\embedded`.

**Download the `ipal-extra` libraries**

These libraries include a TCP/IP stack which is not open-source. Years ago it was stored in https://svn.icub.iit.local, but not anymore. So far it is only stored in our intranet: `\\storage05.icub.iit.local\repository\software\ipal-extra`. As a side note, the batch script `ipal-extra.install.bat` in `<code_root>\icub-firmware\emBODY\eBcode\arch-arm\libs\highlevel\abslayer\ipal` used to retrieve the files from the SVN repo, but now it just tells where to get the extra code.

- copy manually the files from folder `\\storage05.icub.iit.local\repository\software\ipal-extra` into the folder `C:\dev\git\embedded\icub-firmware\emBODY\eBcode\arch-arm\libs\highlevel\abslayer\ipal\ipal-extra`.

**Install the `ipal-extra` libraries**

- in that same folder `C:\dev\git\embedded\icub-firmware\emBODY\eBcode\arch-arm\libs\highlevel\abslayer\ipal`, double-click on `ipal.uvprojx`. This opens the `ipal` library project in the **uVision** IDE,
- build the project (F7).
- Now open the project `C:\dev\git\embedded\icub-firmware\emBODY\eBcode\arch-arm\board\ems004\appl\v2\proj\ems4rd.uvprojx` in the IDE (double-click on the file `ems4rd.uvprojx`), and build (F7).

- ### Marco's comment
the ipal-extra surely contains a TCP/IP stack which is not open-source.
years ago it was stored in https://svn.icub.iit.local, but not anymore. So far it is only stored in our intranet: `\\storage05.icub.iit.local\repository\software\ipal-extra`. the script file hence, dos not retrieve anymore and it just tells where to get the extra code.

isorrentino, 2021-02-06:
### Content of https://github.com/dic-iit/element_torque-control-via-current/issues/126#issuecomment-658288727

#### Steps I followed to configure MPLAB and the compiler

> ## Installing the MPLAB X IDE
> 
> - Goto https://www.microchip.com/mplab/mplab-x-ide and browse down to "Downloads" section and download the latest version of MPLAB® X IDE driver for Windows (currently v5.40).
> - follow the instruction and default options
> 
> ## Installing the MPLAB XC Compilers
> 
> All the compilers and related dependencies can be downloaded from https://www.microchip.com/mplab/compilers, in the last section, in the tab "Compiler Downloads".
> [image]
> 
> The latest version for the compiler and dependencies on Windows is **v1.50**.
> 
> - After the install of the MPLAB X IDE is complete, follow the suggestion to download the "MPLAB XC compiler 16 bits" for dsPICs, and download the "MPLAB XC16 Compiler vX.XX" compiler for Windows, in "Compiler Downloads" tab (as shown above), "Compilers" subsection.
> - Launch the compiler installation: select the "free" option, and select the default options in the steps that follow.
> - Download Install the "MPLAB XC16 Compiler Part-Support Patch vX.XX" from the subsection "Latest Part Support Patch Files". This is a priori for supporting new parts, but actually the dsPIC used in our setup, version **33FJ128MC802**, appears in the "supported parts in the update" secttion of the [release notes of the patch](https://ww1.microchip.com/downloads/en/DeviceDoc/XC16-v1.50-part-support-release-notes.html).
> - Download Install the "Current Peripheral Libraries" -> "16-bit dsPIC33, PIC24E, PIC24H MCUs: Legacy Peripheral Libraries" from the subsection "Peripheral Libraries (PLIBS)" (latest version of "PIC24/dsPIC Peripheral Libraries v2.00 - Windows" [here](https://www.microchip.com/mymicrochip/filehandler.aspx?ddocname=en574967)).
> 
> ### Note on the "MPLAB XC16 Compiler vX.XX" compiler
> 
> The compiler "MPLAB XC16 Compiler v1.50" alone is missing some dependencies like the include `<pwm12.h>`. If you build the project without the mentioned additonnal patches, you'll get the error:
> ```
> CLEAN SUCCESSFUL (total time: 8ms)
> make -f nbproject/Makefile-default.mk SUBPROJECTS= .build-conf
> make[1]: Entering directory 'C:/Users/icub/git/icub-firmware/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/2FOC-V3.X'
> make  -f nbproject/Makefile-default.mk dist/default/production/2FOC-V3.X.production.hex
> make[2]: Entering directory 'C:/Users/icub/git/icub-firmware/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/2FOC-V3.X'
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/2FOC.c  -o build/default/production/_ext/1360937237/2FOC.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/2FOC.o.d"        -omf=elf -DXPRJ_default=default  -legacy-libc    -ffunction-sections -mlarge-data -O3 -I"../include" -msmart-io=1 -Wall -msfr-warn=off    -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/DHES.c  -o build/default/production/_ext/1360937237/DHES.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/DHES.o.d"        -omf=elf -DXPRJ_default=default  -legacy-libc    -ffunction-sections -mlarge-data -O3 -I"../include" -msmart-io=1 -Wall -msfr-warn=off    -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../asm/MeasCurr.s  -o build/default/production/_ext/1360919980/MeasCurr.o  -c -mcpu=33FJ128MC802  -omf=elf -DXPRJ_default=default  -legacy-libc  -I"../asm" -Wa,-MD,"build/default/production/_ext/1360919980/MeasCurr.o.d",--defsym=__MPLAB_BUILD=1,--no-relax  -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/Faults.c  -o build/default/production/_ext/1360937237/Faults.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/Faults.o.d"        -omf=elf -DXPRJ_default=default  -legacy-libc    -ffunction-sections -mlarge-data -O3 -I"../include" -msmart-io=1 -Wall -msfr-warn=off    -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/PWM.c  -o build/default/production/_ext/1360937237/PWM.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/PWM.o.d"        -omf=elf -DXPRJ_default=default  -legacy-libc    -ffunction-sections -mlarge-data -O3 -I"../include" -msmart-io=1 -Wall -msfr-warn=off    -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/ADC.c  -o build/default/production/_ext/1360937237/ADC.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/ADC.o.d"        -omf=elf -DXPRJ_default=default  -legacy-libc    -ffunction-sections -mlarge-data -O3 -I"../include" -msmart-io=1 -Wall -msfr-warn=off    -mdfp="C:/Program Files (x86)/Microchip/MPLABX/v5.40/packs/Microchip/dsPIC33F-GP-MC_DFP/1.2.35/xc16"
> "C:\Program Files\Microchip\xc16\v1.50\bin\xc16-gcc.exe"   ../src/System.c  -o build/default/production/_ext/1360937237/System.o  -c -mcpu=33FJ128MC802  -MMD -MF "build/default/production/_ext/1360937237/Sy [... cut ...]
> - Run app MPLAB X IDE v5.40.
> - Open the project [2FOC-V3.X](https://github.com/robotology/icub-firmware/tree/master/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/2FOC-V3.X) (shortcut CTRL+SHIFT+O on windows).
> - you should see in the lower left pane:
> 
> [image]
> 
> The 2FOC device is the **dsPIC33FJ128MC802**.
> - Then click on Packs - dsPIC33F-GP-...   You should see a popup window with the project properties
> 
> [image]
> 
> just click "OK".
> - Clean and Build the project as shown below:
> [image]

isorrentino, 2021-02-06:
I modified the title of the issue and I summarized the aim in the first comment https://github.com/robotology/icub-firmware/issues/161#issue-797494427.

cc @traversaro @pattacini @maggia80 @marcoaccame @ale-git @DanielePucci @giulioromualdi

pattacini, 2021-02-07:
Thanks @isorrentino for this contribution!

I believe we would should integrate this material into our official documentation at https://icub-tech-iit.github.io/documentation.

cc @marcoaccame

traversaro, 2021-02-12:
fyi @AlexAntn

pattacini, 2021-02-12:
We will be dealing with this next sprint.

traversaro, 2021-02-25:
Related PR: https://github.com/icub-tech-iit/documentation/pull/90 . cc @AlexAntn @isorrentino

pattacini, 2021-02-26:
Documentation ready at See https://icub-tech-iit.github.io/documentation/icub_firmware/fw_toolchain ✔

pattacini, 2021-02-27:
Closing, feel free to open it up again if needed.

closed 2021-02-27 by pattacini