robotology/icub-firmware #164: Discuss the possibility to automatic build the 2FOC firmware with GitHub Action

GiulioRomualdi, 2021-02-01:
This issue follows https://github.com/robotology/icub-firmware/pull/159#issuecomment-770118302. 

**Disclaimer: What I'm presenting here is related to the 2FOC firmware, I don't know if it can be extended to the firmware of other boards**

Even if the code of `icub-firmware` is updated, the new version firmware in `icub-firmware-build` has to be manually pushed, this _manual_ operation can be boring and sometimes the author of the PR may forget to do this. As a consequence, it may happen that a flashed firmware is not related to any commit in `icub-firmware-build`. As a consequence, it might be complex to guarantee the reproducibility of the experiments/demos (please check https://github.com/robotology/community/discussions/450) 

In this issue, I would like to discuss the possibility to automatize the process of deploying the 2FOC firmware.  

Thanks to #163 the firmware compiles on Linux. Thus we may think to use a GitHub action to automatically compile the firmware and push it on `icub-firmware-build`
For instance, [this GitHub action](https://github.com/marketplace/actions/build-with-mplabx-and-xc16) can be used.  

This approach was already employed in [`icub-models-generator`](https://github.com/robotology/icub-models-generator) and [`icub-models`](https://github.com/robotology/icub-models). Indeed the devel branch of `icub-models` is populated by a GitHub Action running on `icub-models-generator`.

I believe that this action will help us (@robotology/iit-dynamic-interaction-control) to better guarantee the repeatability of the software we develop.

cc @DanielePucci @traversaro @isorrentino @marcoaccame @ale-git

pattacini, 2021-02-01:
Hi @GiulioRomualdi 

Thanks for opening this discussion 👍🏻 

Your proposal is very appealing, indeed, and we have it already as an important point on our roadmap 💹
The roadmap foresees a change of our whole toolchain ⚙ to head toward more open and more flexible methods that will in turn integrate tightly with the GitHub actions stack.

@marcoaccame is already working out alternative toolchains with these perks in mind, although this change won't take place soon.

Regarding the 2FOC, I think it's a doable task. We'd only need to plan for it.

traversaro, 2021-02-01:
Note that two important aspects that to enable GitHub Actions enabled build are:
* Support headless (i.e. from command line) installation of the build
* Support headless (i.e. from command line) compilation of the firmware

pattacini, 2021-02-01:
Putting @Nicogene in the loop as he was interested in this as well.

GiulioRomualdi, 2021-02-01:
Hi @traversaro it's possible to compile the firmware of the 2foc from the terminal (Linux)
I tried on my laptop 
Please check here: https://github.com/velocitek/ghactions-mplabx/blob/c59aac181bc581925b0d34ed218fd89527edfa6f/build.sh#L7-L8

traversaro, 2021-02-01:
> 
> 
> Hi @traversaro it's possible to compile the firmware of the 2foc from the terminal (Linux)
> I tried on my laptop
> Please check here: https://github.com/velocitek/ghactions-mplabx/blob/c59aac181bc581925b0d34ed218fd89527edfa6f/build.sh#L7-L8

And from the same repo also the first condition seems to be satisfied: https://github.com/velocitek/ghactions-mplabx/blob/c59aac181bc581925b0d34ed218fd89527edfa6f/Dockerfile#L3 .