robotology/icub-firmware #165: Understand why current shown on power supply and robot displays is different from the current read through WBToolbox

isorrentino, 2021-02-08:
Hi all, I do not know if this is where this issue had to be opened. Here the problem.

While checking the current of the `left ankle roll motor` with @MrAndrea and @giulioromualdi after https://github.com/robotology/icub-tech-support/issues/1070, we noticed that the total current required by the robot shown on the power supply display and the robot display was lower than the current read by using WBToolbox:

[image]

Indeed, here is a plot of what has been measured from Simulink

[image]

The plot shows the current of the `left ankle roll motor` and is expressed in Ampere. While the plot says that the motor was asking for a current between 10 and 12 Ampere the power supply reached a maximum value of 10/11 Ampere, where this value is the current needed to supply the entire robot. It seems that something is not matching.

Here https://github.com/robotology/icub-tech-support/issues/1070#issuecomment-773945305 the comment of @MrAndrea (that's why I'm opening this issue). Maybe as soon as the robot is back (see https://github.com/icub-tech-iit/task-force-miscellanea/issues/69#issuecomment-771722284) the problem can be double-checked.

cc @pattacini @traversaro @ale-git @DanielePucci @S-Dafarra

S-Dafarra, 2021-02-08:
Regarding the difference, this comment by @maggia80 might be relevant: https://github.com/icub-tech-iit/task-force-miscellanea/issues/58#issuecomment-738343085
> consider that the current of the motor is not directly correlated to the current of the power source. There is an "inverter" in between. You need just few volts to apply 10A to the motor in quasi-static condition -> you need about 8V to apply 10A to the motor windings, while the power source will see only 2A @40V.

This is the code that reads the currents in ``WBToolbox``: https://github.com/robotology/wb-toolbox/blob/99156c73c6b651502317722dc908ba7a9ba5f4ce/toolbox/library/src/GetMeasurement.cpp#L314-L320

ale-git, 2021-02-08:
Yes, exactly.

isorrentino, 2021-02-08:
Great. So, if the comment https://github.com/icub-tech-iit/task-force-miscellanea/issues/58#issuecomment-738343085 explains the behaviour we can also close this issue. Do you agree?

pattacini, 2021-02-08:
Closing, feel free to open it up again if needed.

closed 2021-02-08 by pattacini