robotology/icub-firmware #172: Revision of the development toolchain for ARM-based boards

marcoaccame, 2021-02-22:
The toolchain used for the development of the embedded software for the ARM-based boards is at the moment based on the following tools:

- The creation of a low level HW driver is done by means of `CUBE-MX` by STMicroelectronics. We use MPUs only by STM and the use of this tools is straightforward.
- The IDE and compiler is the `Keil MDK ARM`. With this one we manage projects which organize the required code in static libraries and produce the executables which run on the boards. These latter are `bootloader` / `application` for the CAN boards or `eLoader` / `eUpdater` / `eApplication` for the ETH boards. 
- We have templates for the quick creation of typical executable with sample projects and associated startup and scatter files to load execution at a particular FLASH address.
- We use the `ULINK-Pro` and the `STLINK` debugger to run and trace code on the target.

These tools are consolidated and have surely served well over the years, but nevertheless we would like to investigate others that can have some desirable features.

This issue is a placeholder of such an investigation.

We would like to explore other compilers  (gcc / g++ for instance) and build / debug strategies which satisfy at least the following, possibly with different weights:

- Free access to the compiler without need to purchase a license.
- State of the art IDE (integrated Development Environment) which let us navigate modern C++.
- Possibility to automate the compilation process,
- Have satisfactory debug capabilities using at least the STLIMK debugger (but the reuse of ULINK-Pro and of others is desirable),
- Have CMake integration
- Keep our code infrastructure, possibly with minor changes. For code infrastructure I mean the use of static libraries which holds similar services (such as low level HW driver, RTOS, communication stack, ...) and C and C++ shared files all organized for the creation of executables and templates projects for those executable.
- Possibility to uses C / C++ code generated by model-based design, e.g. by Simulink.

maggia80, 2021-03-02:
cc: @pattacini @traversaro @vtikha

maggia80, 2021-03-02:
https://medium.com/@lixis630/getting-started-to-code-embedded-c-on-stm32-e90e7910b2c

GiulioRomualdi, 2021-03-02:
@isorrentino

maggia80, 2021-03-10:
interesting overview [stm32-stm8_embedded_software_solutions.pdf](https://github.com/robotology/icub-firmware/files/6114112/stm32-stm8_embedded_software_solutions.pdf)