robotology/icub-firmware #190: Document data types 2FOC

giovannipizzolante, 2021-06-15:
By looking at the documentation of the XC compiler 16 (https://ww1.microchip.com/downloads/en/DeviceDoc/MPLAB_XC16_C_Compiler_Users_Guide_50002071L.pdf), me and @GiulioRomualdi noticed that the `short` data types seems to be 2 bytes long as the `int` data type. As shown in the following table. 

[image]

Is this correct? Furthermore, is this the correct documentation?

pattacini, 2021-06-15:
cc @ale-git

ale-git, 2021-06-15:
Yes, it is correct: the **int** type is 16 bit.

giovannipizzolante, 2021-06-15:
and the **short** type?

ale-git, 2021-06-15:
The same, 16 bit like in your table.

closed 2021-06-15 by pattacini

GiulioRomualdi, 2021-06-15:
cc @isorrentino

giovannipizzolante, 2021-07-05:
The User's Guide in the first comment is wrong. The correct one is this one:

[User'sGuide.pdf](https://github.com/robotology/icub-firmware/files/6763434/User.sGuide.pdf)

cc @isorrentino