robotology/icub-firmware #270: Implement a temporary check to verify the frequency at which ems4 and mc4plus stream data

gabrielenava, 2022-05-24:
During recent experiments with iRonCub we tried to increase the frequency of ems4 and mc4plus board, however it was hard to verify if the boards actually stream data at the desired frequency. 

Talking to @marcoaccame we realized that there is a quick and dirty solution to check if the frequency is correct, that is, stream also the time stamp at which data are send by the board. To avoid big modifications of the YARP interface we thought to use the `joints acceleration` slot in `stateExt` port to temporarily stream this data and verify that the desired frequency is reached.

cc @pattacini @DanielePucci @maggia80

marcoaccame, 2022-05-24:
> Talking to @marcoaccame we realized that there is a quick and dirty solution to check if the frequency is correct, that is, stream also the time stamp at which data are send by the board. To avoid [...]

The idea is to put something else inside the variable which reports the acceleration of the joint, for instance the system time of the ETH board expressed in ms.  That can be done by changing the following [code](https://github.com/robotology/icub-firmware/blob/devel/emBODY/eBcode/arch-arm/embobj/plus/mc/Joint.c#L927-L936) as such:

```diff
+#include "EOVtheSystem.h"
void Joint_get_state(Joint* o, eOmc_joint_status_t* joint_state)
{
    joint_state->core.modes.interactionmodestatus    = o->interaction_mode;
    joint_state->core.modes.controlmodestatus        = o->control_mode;
    joint_state->core.modes.ismotiondone             = Trajectory_is_done(&o->trajectory);
    joint_state->core.measures.meas_position         = o->pos_fbk;           
    joint_state->core.measures.meas_velocity         = o->vel_fbk;        
+   joint_state->core.measures.meas_acceleration     = eov_sys_LifeTimeGet(eov_sys_GetHandle()) / 1000;      
-   joint_state->core.measures.meas_acceleration     = o->acc_fbk;
    joint_state->core.measures.meas_torque           = o->trq_fbk;    
}
``` 
That must be done by changing and recompiling the same code of icub-firmware and icub-firmware-shared used to produce the same binary running on the robot. I would recommend to align the robot to the latest devel.

Also, it is handy to use a dedicated application version by changing the following [code ](https://github.com/robotology/icub-firmware/blob/devel/emBODY/eBcode/arch-arm/board/ems004/appl/v2/cfg/eoemsappl/EOMtheEMSapplCfg_cfg.h#L81-L84) as well:  

```diff
#define EOMTHEEMSAPPLCFG_VERSION_MAJOR          (VERSION_MAJOR_OFFSET+3)
//  <o> minor           <0-255> 
//  <o> minor           <0-255>
-#define EOMTHEEMSAPPLCFG_VERSION_MINOR          51
+#define EOMTHEEMSAPPLCFG_VERSION_MINOR         151
```

If the reading from teh relevant YARP port gives a unitary variation, then it means that the values at the joints are successfuly propagated every 1 ms.

marcoaccame, 2022-06-17:
Hi @gabrielenava ,
here are the binaries to use in which I just did the changes in https://github.com/robotology/icub-firmware/issues/270#issuecomment-1136116393 to bot the `ems` and `mc4plus`:
- [ems.v3.151-meas_acceleration-has-millisec.zip](https://github.com/robotology/icub-firmware/files/8926214/ems.v3.151-meas_acceleration-has-millisec.zip)
- [mc4plus.v3.149-meas_acceleration-has-millisec.zip](https://github.com/robotology/icub-firmware/files/8926232/mc4plus.v3.149-meas_acceleration-has-millisec.zip)

gabrielenava, 2022-06-20:
cc @fabioBergonti

FabioBergonti, 2022-06-22:
Hi @marcoaccame 
I and @antonellopaolino are testing your modification. 
What's the unit of measurement of the timestamp? [s] or [das]?

I dumped `meas_acceleration` with simulink and here the result

[image]

marcoaccame, 2022-06-23:
milli seconds

from: https://github.com/robotology/icub-firmware/issues/270#issuecomment-1136116393

> The idea is to put something else inside the variable which reports the acceleration of the joint, for instance the system time of the ETH board expressed in ms.  That can be done by changing the [...]

marcoaccame, 2022-06-23:
**NOTE**: if you read the value from a `YARP` port you should also verify if and how `YARP` transforms the `meas_acceleration` variable before it publishes it to the relevant port. This variable is natively an integer where I put ms values, but `YARP` may transform it.

If it can help, this variable (when not hacked) contains acceleration expressed in `idegrees/s^2`, and and `1 degree = 182.0444444444444 idegrees`.

Once you have applied the reverse transformation which `YARP` apply you shall have a log vector of integer values holding ms values which must be monotonically increasing.

Now, the ETH board transmits inside the UDP frame the value in ms from the boostrap. If you have configured to transmit one frame every 1 ms, to receive inside linux w/ a delta on 1 ms and configured the wrapper to publish every 1 ms (and I am sure that is already OK)  and you read the YARP port at exactly 1 ms delta, then you should have a vector v[] of integer values which:
- is monotonically increasing,
- follows the rule v[n+1] = v[n]+1.

To verify that there are no holes (which means no value is lost) you should:
- plot v[n+1] - v[n]
- verify that the plot contains only 1 values. 

As an alternative which does not use the reverse transformation just plot v[n+1] - v[n] and verify that it is constant

FabioBergonti, 2022-06-23:
> To verify that there are no holes (which means no value is lost) you should:
> 
>     * plot v[n+1] - v[n]
> 
>     * verify that the plot contains only 1 values.

Here is the result
(v[n+1] - v[n]) is constant expect for some periodic spikes
- using simulink
[image]
- via `yarp read ... /icub/head/stateExt:o >> data.log`
[image]

marcoaccame, 2022-06-23:
I suspect that the value you plot have some missed values. Where are they lost?

Pls look at the log of yarprobotinterface. If you don't see any message emitted by the following code, then it may be that YARP or the scheduling in icub-head or in the client PC which reads the port may miss values.
```C++
void cpp_protocol_callback_incaseoferror_in_sequencenumberReceived(EOreceiver *r)
{  
    const eOreceiver_seqnum_error_t * err = eo_receiver_GetSequenceNumberError(r);
    long long unsigned int exp = err->exp_seqnum;
    long long unsigned int rec = err->rec_seqnum;
    long long unsigned int timeoftxofcurrent = err->timeoftxofcurrent;
    long long unsigned int timeoftxofprevious = err->timeoftxofprevious;
    char *ipaddr = (char*)&err->remipv4addr;
    //printf("\nERROR in sequence number from IP = %d.%d.%d.%d\t Expected: \t%llu,\t received: \t%llu\n", ipaddr[0], ipaddr[1], ipaddr[2], ipaddr[3], exp, rec);
    char errmsg[256] = {0};
    snprintf(errmsg, sizeof(errmsg), "hostTransceiver()::parse() detected an ERROR in sequence number from IP = %d.%d.%d.%d. Expected: %llu, Received: %llu, Missing: %llu, Prev Frame TX at %llu us, This Frame TX at %llu us",
                                    ipaddr[0], ipaddr[1], ipaddr[2], ipaddr[3],
                                    exp, rec, rec-exp,
                                    timeoftxofprevious, timeoftxofcurrent);
    yError() << errmsg;
}
``` 
**Code Listing.** From file `hostTransceiver,cpp` here is the handler which detects missing UDP frames. If this code does not execute, we have the certainty that all the information is delivered to YARP.

FabioBergonti, 2022-06-28:
We are working at 500Hz
I've repeated another test where I dumped `/icub/head/stateExt:o` and I saved the output of the yarplogger.

Here is the `v[n+1] - v[n]`:
[image]

Here is the output of the yarplogger:
``` log
 24.437725 <DEBUG> yarprobotinterface running happily
 51.116856 <INFO>  from BOARD 10.0.1.20 (head-eb20-j0_1), src LOCAL, adr 0, time 1146s 938m 13u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 51.218195 <INFO>  from BOARD 10.0.1.21 (head-eb21-j2_5), src LOCAL, adr 0, time 1147s 38m 311u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 58.746323 <INFO> Sending output from /icub/head/stateExt:o to /tmp/port/1 using tcp
 68.030576 <WARNING> xsensmt: Missing orientation message, skipping orientation update.
 68.030714 <WARNING> xsensmt: Missing accelerometer message, skipping accelerometer update.
 68.030756 <WARNING> xsensmt: Missing gyroscope message, skipping gyroscope update.
 68.040697 <WARNING> xsensmt: Missing magnetometer message, skipping magnetometer update.
 81.614021 <WARNING> xsensmt: Missing orientation message, skipping orientation update.
 81.614149 <WARNING> xsensmt: Missing accelerometer message, skipping accelerometer update.
 81.614196 <WARNING> xsensmt: Missing gyroscope message, skipping gyroscope update.
 81.614232 <WARNING> xsensmt: Missing magnetometer message, skipping magnetometer update.
 84.436117 <DEBUG> yarprobotinterface running happily
 104.458002 <INFO>  from BOARD 10.0.1.7 (left_leg-eb7-j4_5), src LOCAL, adr 0, time 1200s 38m 753u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.458107 <INFO>  from BOARD 10.0.1.1 (left_arm-eb1-j0_3), src LOCAL, adr 0, time 1200s 40m 145u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.458149 <INFO>  from BOARD 10.0.1.6 (left_leg-eb6-j0_3), src LOCAL, adr 0, time 1200s 39m 112u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.468120 <INFO>  from BOARD 10.0.1.3 (right_arm-eb3-j0_3), src LOCAL, adr 0, time 1200s 40m 356u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.468262 <INFO>  from BOARD 10.0.1.9 (right_leg-eb9-j4_5), src LOCAL, adr 0, time 1200s 38m 239u: (code 0x06000005, par16 0x0005 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.569309 <INFO>  from BOARD 10.0.1.5 (torso-eb5-j0_2), src LOCAL, adr 0, time 1200s 138m 896u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 104.579510 <INFO>  from BOARD 10.0.1.8 (right_leg-eb8-j0_3), src LOCAL, adr 0, time 1200s 139m 399u: (code 0x06000005, par16 0x0007 par64 0x3000000000000000) -> ETH monitor: just verified, no news + .
 115.056305 <INFO> Removing output from /icub/head/stateExt:o to /tmp/port/1
 120.834970 <WARNING> xsensmt: Missing orientation message, skipping orientation update.
 120.835047 <WARNING> xsensmt: Missing accelerometer message, skipping accelerometer update.
 120.835075 <WARNING> xsensmt: Missing gyroscope message, skipping gyroscope update.
 120.835098 <WARNING> xsensmt: Missing magnetometer message, skipping magnetometer update.
 121.787173 <WARNING> xsensmt: Missing orientation message, skipping orientation update.
 121.787281 <WARNING> xsensmt: Missing accelerometer message, skipping accelerometer update.
 121.787337 <WARNING> xsensmt: Missing gyroscope message, skipping gyroscope update.
 121.787383 <WARNING> xsensmt: Missing magnetometer message, skipping magnetometer update.
 144.440557 <DEBUG> yarprobotinterface running happily
```

marcoaccame, 2022-06-28:
I read the above as follows:
- the log shows no lost packet, so `embObjMotionControl` receives every UDP packet w/ time updated at steps of 2 ms. 
- the value in ms is published by the YARP port and read by some external agent
- this agent receives consecutive values w/ a delta which
   - mostly is =~0.011. I assume that this value corresponds to the expected delta of 2 ms.
   - sometimes it is = 0. Two consecutive readings are equal. 
   - some other times it is = 2*0.011. 

My interpretation is that that reading agent and the publishing YARP port are mostly synchronized apart from a few times when two consecutive readings are equal or one is missed.

So, not that bad.

pattacini, 2022-07-09:
Closing.
Feel free to open it up again if need more inquiring.

closed 2022-07-09 by pattacini