robotology/icub-firmware #360: AMC-BLDC: implement concistency check in the HALL configurations.

valegagge, 2023-03-27:
We want to implement the consistency check in the HALL configurations that detects impossible transitions as the 2FOC fw already does.

**Dod**
The check is implemented and tested.
The related PR has been merged.

MrAndrea, 2023-04-03:
We have to check if there are already messages for the overcurrent and fault status.

valegagge, 2023-04-03:
> We have to check if there are already messages for the overcurrent and fault status.

In detail: we want to check for other faults at HAL level to propagate to the user.

ale-git, 2023-04-14:
I'm implementing the same consistency check (HALL allowed state and allowed transitions) used in the 2FOC firmware, work in progress.

ale-git, 2023-05-11:
https://github.com/robotology/icub-firmware/pull/378

I've added hall sensor effect state and sequence check logic in order to detect errors. It provides functions to check value and sequence error state in the hall sensors and overcurrent state, that was already detected but not reported by the HAL. It takes no action that is demanded to the supervisor. Internally it uses a 32 bit faultmask inherited from the 2FOC protocol. Not all error conditions are meaningful within the AMC-BLDC environment, anyway the 32 bit mask can be made available to replicate the CAN state message of the 2FOC.

valegagge, 2023-06-08:
The PR #378 has been merged. Closing issue...

closed 2023-06-08 by valegagge