robotology/icub-firmware #396: Friction model coulomb + viscous cannot be used without rebuilding the firmware

isorrentino, 2023-07-16:
Hi all, I started to use the coulomb + viscous model for the friction compensation on ergoCub. In order to use it I had to rebuild the firmware and flash the boards of the robot. It would be nice to have the possibility to choose it without rebuilding the firmware.

pattacini, 2023-07-17:
@sgiraz, would you mind taking care of this?

sgiraz, 2023-07-18:
Hi guys,

Yes I can work on it starting from the next sprint (Monday 24th)

pattacini, 2023-07-31:
A quick update on this activity.

As we aim to streamline the code and get only one FW capable of dealing with the entire pool of robots, we foresee to:
- Remove the define [`USE_VISCOUS_COULOMB`][1]. As a result, the current law will be replaced by the latest policy outright. We verified that the current law can be subsumed by the new law when default params are employed.
- Make [`threshold`][2] configurable from XML files. While doing so, we check if we can stick to `icubdeg/s` or `deg/s` units.
- Update `ergoCub` XML files to set `threshold` equal to the value that is currently hand-coded in the FW.
- Update the robots that make use of `Kbemf` (there are a few) so that they'll rely on the new friction parameters.

This activity will commence this week and will be most likely completed the next sprint.

[1]: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/embobj/plus/mc/Pid.c#L199
[2]: https://github.com/robotology/icub-firmware/blob/master/emBODY/eBcode/arch-arm/embobj/plus/mc/Pid.c#L212

sgiraz, 2023-08-23:
Hi guys, 

I have some updates about this issue.

### Configuration file changes

Here is a snippet example of the configuration file I used to test the new `threshold` parameter on a single joint setup (`aea2-setup`) + `mc4plus`. Note that I removed the `kbemf` parameter

```diff
    <!-- custom PIDs: begin -->
     <group name="TRQ_PID_OUTPUT_CURR">
        <param name="controlLaw">          torque           </param>
        <param name="outputType">          current          </param>
        <param name="fbkControlUnits">     metric_units     </param>
        <param name="outputControlUnits">  machine_units    </param>
        <param name="kp">                  0                </param>
        <param name="kd">                  0                </param>
        <param name="ki">                  0                </param>
        <param name="maxOutput">           2500             </param>
        <param name="maxInt">              200              </param>
        <param name="ko">                  0                </param>
        <param name="stictionUp">          0                </param>
        <param name="stictionDown">        0                </param>
        <param name="viscousPos">          96                </param>
        <param name="viscousNeg">          97                </param>
        <param name="coulombPos">          98                </param>
        <param name="coulombNeg">          99                </param>
+        <param name="threshold">           11                </param>
        <param name="kff">                 1                </param>
-        <param name="kbemf">               0.00             </param>
        <param name="filterType">          0                </param>
        <param name="ktau">               50.0              </param>
    </group>
```

### icub-main: eomcParser

After the required changes, as we can see from the image below the `eomcParser` is now able to parse the new `threshold` parameter.

[image]

### icub-firmware-shared: 

A new threshold parameter has been added to the `eOmc_FrictionParams_t` structure. This required 4 Bytes more in the ROP size.  

```diff
typedef struct
{
    float32_t               viscous_pos_val;
    float32_t               viscous_neg_val;
    float32_t               coulomb_pos_val;
    float32_t               coulomb_neg_val;
+    float32_t               threshold_val;
} eOmc_FrictionParams_t;    EO_VERIFYsizeof(eOmc_FrictionParams_t, 20)
```

### icub-firmware:

The new parameter reaches correctly the FW level:

[image]

---

What to do now:
- Test the SW+FW on a complete setup (ems+2foc + bldc motor) / real robot, where we can try to switch in `torque-control-mode` and see if it works as expected.
- Update the configuration files of the robots paying attention the the ones that may use the `kbemf` parameter
- Draft all the PRs

cc @marcoaccame @pattacini @isorrentino

pattacini, 2023-08-24:
Nice!

Now that I look at the table of parameters I feel that `threshold` may be a too anonymous name. Better off something like `velocityThres`. What do you think @sgiraz?

Also, we should check the units of the new params ahead of the tests.

sgiraz, 2023-08-28:
Hi @pattacini,

I agree with you, I just changed the name of the variable to `verlocityThres`.
Accordingly with the caller [function][1] we can see that the units for the new parameter must be `icubdeg/s`.

Now, I proceed to remove the old `kbemf` parameter from all the robots in robots-configuration, introducing the new `velocityThresh`.

[1]: https://github.com/robotology/icub-firmware/blob/9240bb9927dbba613d15b5e7cb697a78e22d3d98/emBODY/eBcode/arch-arm/embobj/plus/mc/Motor.c#L903

cc @isorrentino

pattacini, 2023-08-28:
> Looking at the [caller function](https://github.com/robotology/icub-firmware/blob/9240bb9927dbba613d15b5e7cb697a78e22d3d98/emBODY/eBcode/arch-arm/embobj/plus/mc/Motor.c#L903) we can see that the units for the new parameter must be `icubdeg/s`.

This doesn't necessarily imply that we must stick to `icubdeg/s`.
For example, we can switch to `deg/s` units in the conf files and then do the conversion in the FW.
Of course, switching to `deg/s` is only helpful if it allows us to keep homogeneity with other parameters, for instance. Is this the case?

> [!important]
Also, once we pick the units, we need to check whether typical user values for this parameter can be transmitted with the least loss of precision.

In this latter respect, `icubdeg/s` requires higher values than `deg/s` for the same configuration, if I'm not mistaken. Therefore, choosing units has an impact also on the precision we can convey on the net.

sgiraz, 2023-08-28:
> Important
> Also, once we pick the units, we need to check whether typical user values for this parameter can be transmitted with the least loss of precision.

Hi @isorrentino, 

Do you believe that We can use `deg/s` instead of `icubdeg/s` considering the note highlighted by @pattacini?

>**Note**
>Consider that the `velocityThresh` parameter is declared as [float32_t][1]. Therefore, the [FLT_EPSILON][2] = 1.192093e-07.

[1]: https://en.cppreference.com/w/cpp/types/floating-point 
[2]: https://en.cppreference.com/w/c/types/limits

valegagge, 2023-08-29:
Hi @sgiraz ,
just a question: did you update the `icub-templates` files accordingly with your changes?

Remember that if your parameter is not mandatory, you should not increment the version. 

Thanks!

sgiraz, 2023-08-29:
Hi @valegagge,

I haven't changed the `robots-configuration` yet, but I'll take into account your suggestions.
As of now, I identified the following conditions:

<table>
<tr><th>Not in Icebox </th><th>In Icebox</th></tr>
<tr><td>

| robot                          | condition of kbemf param |
| :----------------------: | :----------------------------: |
| ergoCubSN000          | ⚠️‼️ |
| ergoCubSN001            |⚠️‼️  |
| iCubEdinburg01           | ✅  |
| iCubErzelli01                | ✅  |
| iCubErzelli02                | ✅  |
| iCubGenova02             | ⚠️❗   |
| iCubGenova07             | ⚠️ |      
| iCubGenova08             | ⚠️ |
| iCubGenova10             | ⚠️ |
| iCubGenova11             | ⚠️ |
| iCubLisboa01               | ⚠️ |
| iCubNancy01               | ✅ |
| iCubPargue01              | ⚠️ |
| iCubShanghai01          | ⚠️ |
| iCubShenzhen01         | ⚠️ |
| iCubSingapore01        | ✅ |
| iCubTemplates            | ✅ |
| iCubValparaiso01        | ⚠️ |
| iCubWaterloo01          |⚠️ |
| iCubZagreb01             |⚠️ |

</td><td>

| robot                          | condition of kbemf param |
| :----------------------: | :----------------------------: |
| iCubBarcelona         |    ✅    |
| iCubChemnitz          |    ⚠️❗  |
| iCubDarmstadt         |    ✅    |
| iCubGenova04          |    ⚠️‼️   |
| iCubGenova09          |    ⚠️‼️   |
| iCubHeidelberg01      |    ✅    |
| iCubHongKong01        |    ⚠️    |
| iCubLausanne02        |    ✅    |
| iCubMoscow01          |    ⚠️    |
| iCubNottingham01      |    ✅    |
| iCubShieffield01      |    ✅    |
| iCubTwente01          |    ✅    |

</td></tr> </table>

Where:
- ✅ means that the `kbmef` exists but is 0 on all the joints
- ⚠️ means that the `kbmef` exists and is not 0 on the following joints: [left_arm-eb1-j0_3][1], [right_arm-eb3-j0_3][2], [torso-eb5-j0_2][3]
- ⚠️❗ means that the `kbmef` exists but is not 0 on the following joints: left_arm-eb1-j0_3, right_arm-eb3-j0_3, torso-eb5-j0_2, left_leg-eb6-j0_3, right_leg-eb8-j0_3
- ⚠️‼️ means that the `kbmef` exists but is not 0 on a significant number of joints

[1]: https://github.com/robotology/robots-configuration/blob/100b381de92b0b13cd3348c87f13c98268e5441e/iCubZagreb01/hardware/motorControl/left_arm-eb1-j0_3-mc.xml#L79
[2]: https://github.com/robotology/robots-configuration/blob/100b381de92b0b13cd3348c87f13c98268e5441e/iCubZagreb01/hardware/motorControl/right_arm-eb3-j0_3-mc.xml#L79
[3]: https://github.com/robotology/robots-configuration/blob/100b381de92b0b13cd3348c87f13c98268e5441e/iCubZagreb01/hardware/motorControl/torso-eb5-j0_2-mc.xml#L79

>**Note**
> - All the robots with the `kbemf` parameter in the (⚠️) state have the same values on the same joints. As a result, I guess most of them are cut-and-pasted.
> - When `kbemf` is 0, it's like it doesn't exist in the friction compensation control law
> - Surely, the new parameter needs to be fine-tuned for ergoCub robots

cc @isorrentino @pattacini @Nicogene

sgiraz, 2023-08-29:
PRs drafted:
- https://github.com/robotology/yarp/pull/3008
- https://github.com/robotology/icub-main/pull/902
- https://github.com/robotology/icub-firmware-shared/pull/86
- https://github.com/robotology/icub-firmware/pull/410
- https://github.com/robotology/icub-firmware-build/pull/100

TODO:
- < robots-configuration > 

cc @isorrentino @pattacini

sgiraz, 2023-09-01:
Since we decided to add the new `velocityThres` parameter without remove the old `kbemf` one it is not necessary to update the robots-configurations for all the robots.

Before merging all the repos, I would ask to @isorrentino to give a try on `ergoCub1.0 S/N:000` providing us some feedbacks.

cc @Nicogene @pattacini @marcoaccame

pattacini, 2023-09-01:
> Since we decided to add the new `velocityThres` parameter without remove the old `kbemf` one it is not necessary to update the robots-configurations for all the robots.

`kbemf` remains in YARP but the FW no longer employs it, I think. Therefore, we still need to update the robot conf files.

Remember that robots located in icebox will be always untouched.

sgiraz, 2023-09-01:
/remind 18 Sep Let's try the new parameter on `ergoCub1.0 S/N:000` (cc @isorrentino @Nicogene @pattacini)

valegagge, 2023-09-06:
> > Since we decided to add the new `velocityThres` parameter without remove the old `kbemf` one it is not necessary to update the robots-configurations for all the robots.
> 
> `kbemf` remains in YARP but the FW no longer employs it, I think. Therefore, we still need to update the robot conf files to use the new param.
> 
> Remember that robots located in icebox will always stay unchanged.

I think that leaving the old parameter might be confusing on the user side...
if all the sw constraints for back compatibility are respected, I suppose that we should not face any problems....

pattacini, 2023-09-06:
> I think that leaving the old parameter might be confusing on the user side...

I agree 👍🏻 
We ought to update robots' conf files to the use of the new param straight away, even though YARP will still contain the machinery to deal with the old param.

valegagge, 2023-09-06:
@sgiraz 
Please also update the icub-templates config file with the new param.
If you need more information please ask me. 😄

pattacini, 2023-09-20:
Most likely to postpone to the next week as ergoCub is going to showcase during IIT anniversary tomorrow.

/remind September 26 Let's try the new parameter on ergoCub1.0 S/N:000 (cc @isorrentino @sgiraz @Nicogene)

sgiraz, 2023-09-28:
Hi @isorrentino,

When you are ready, you can start the tests on this activity. Let us know, please.

pattacini, 2023-10-05:
Hi @isorrentino 

@sgiraz is sick and thus unavailable for a while. This could potentially slow down the tests, but no problem in that sense.

The message you reported seems to indicate a mismatch between the FW and the SW.
I'd need to check it w/ the help of @marcoaccame.

For confirmation, are you also using the `icub-main` and `yarp` versions reported in the [list above](https://github.com/robotology/icub-firmware/issues/396#issuecomment-1697219349)?

isorrentino, 2023-10-05:
> For confirmation, are you also using the icub-main and yarp versions reported in the https://github.com/robotology/icub-firmware/issues/396#issuecomment-1697219349?

You are right, I did not recompile `yarp` and `icub-main`. So, I do not know if recompiling `yarp` and `icub-main` things work flashing only one board.

pattacini, 2023-10-05:
>So, I do not know if recompiling yarp and icub-main things work flashing only one board.

Perhaps, the best option would be to flash the boards of one entire leg and launch that part only from the `yarprobotinterface`. 

Just to get aligned on the things to use:
- The FW binaries need to be fetched from @sgiraz's fork (https://github.com/robotology/icub-firmware-build/pull/100) as in the meanwhile the upstream repo has certainly progressed.
- [PR](https://github.com/robotology/yarp/pull/3010) on `YARP` got merged so you can refer to the upstream repo `master` branch.
- By contrast, `icub-main` and `icub-firmware-shared` versions need to be fetched from @sgiraz's forks:
    - https://github.com/robotology/icub-firmware-shared/pull/86
    - https://github.com/robotology/icub-main/pull/903

If this test fails for whatever reason, we can wait for @sgirar.

isorrentino, 2023-10-05:
Thank you for the recap. I hope we already have the config file to run only one leg. I'll test this as soon as I have a new slot on the robot.

isorrentino, 2023-10-11:
Today, I flashed e rebuilt everything reported in https://github.com/robotology/icub-firmware/issues/396#issuecomment-1748331088. 
I use this [branch](https://github.com/ami-iit/robots-configuration/tree/testVelocityThres) for `robots-configuration`. 

Flashing only the right leg as suggested in https://github.com/robotology/icub-firmware/issues/396#issuecomment-1748331088

> Perhaps, the best option would be to flash the boards of one entire leg and launch that part only from the yarprobotinterface.

does not allow to test the friction compensation. Indeed, if we do not run the full robot, we do not have `WBD` running up and as a consequence we do not have the torque feedback.

Furthermore, I do not see the new parameter `velocityThres` from the motorgui.

cc @pattacini @sgiraz

pattacini, 2023-10-11:
Hi @isorrentino 

@sgiraz has recovered and can now follow up on this.

sgiraz, 2023-10-11:
Hi @isorrentino,

> Furthermore, I do not see the new parameter velocityThres from the motorgui.
 
The change has been applied in https://github.com/sgiraz/yarp/blob/81b7f3c0a52f574912bf6c953f927a8e2710b1c7/src/yarpmotorgui/piddlg.ui#L939-L943

Anyway, tomorrow I'll try to verify if there is some kind of mistake about it.

> I tried to flash all the EMS boards and start the robot with the usual [config file](https://github.com/ami-iit/robots-configuration/blob/testVelocityThres/ergoCubSN000/ergocub_wbd.xml) but there are several errors due to incompatible firmware versions.

As indicated in the log posted, there is a misalignment between the `icub-fw-shared` used to build the `FW` and the one used to build `icub-main`.

```console
 2019,134514 <ERROR> EthResource::verifyEPprotocol() for ep = eoprot_endpoint_motioncontrol detected: pc104.version.minor = 26 and board.version.minor = 25
 2019,134519 <ERROR> ACTION REQUIRED: BOARD head-eb20-j0_1 with IP 10.0.1.20 needs a FW update to offer services for eoprot_endpoint_motioncontrol
```

To be sure, I can try to regenerate and provide you with the binaries for all the ETH boards. Meanwhile, I recommend verifying that you have the correct versions of `icub-firmware-shared` and `icub-main` installed.

sgiraz, 2023-10-18:
Hi @isorrentino,

Today I found a bug in `yarpmotorgui` that was causing the crash (segmentation fault) when the `PID` button was clicked independently from the control mode currently set.
As you can see from the picture below, now it works as expected:

[image]

I just dropped a new PR in `yarp`:
- https://github.com/robotology/yarp/pull/3030

You can wait the merge or use my fork for your tests.

Anyway, I also regenerated all the binaries for the ETH boards as mentioned above. See the PR updated:
- https://github.com/robotology/icub-firmware-build/pull/100

cc @isorrentino @pattacini @Nicogene

pattacini, 2023-10-19:
>Unfortunately I cannot test the torque control mode with our setup due to a missing HW configuration.

So, it seems that it's difficult to test the torque mode in action on a setup, while we tested that the information is delivered to the FW.

>By the way, you shouldn't have any problem on the robot. Let me know.

This could be problematic these days, given the robot agenda.
Probably, we need to wait for ergoCubSN001 to be ready.

sgiraz, 2023-10-26:
Hi guys,

I want to drop here just a quick update.
After a quick F2F meet-up with @valegagge, I discovered that the error mentioned in the [comment above][1]:

[image]

is caused by a preliminary check at FW level from the `ems` board:

```c
    if (WatchDog_check_expired(&o->trq_fbk_wdog))
    {
        o->trq_fbk = ZERO;
        o->trq_ref = ZERO;
        o->trq_err = ZERO;
        
        if (o->trq_control_active)
        {
            o->fault_state.bits.torque_sensor_timeout = TRUE;
            
            o->control_mode = eomc_controlmode_hwFault;
        }
    }
```

The watchdog on the `trq_fbk_wdog` is set with a timeout of 100 milliseconds in Joint.c (init function)

```c
   WatchDog_set_base_time_msec(&o->trq_fbk_wdog, 2*DEFAULT_WATCHDOG_TIME_MSEC);
```

And it expires because no feedback is provided by the motion controller

```c
void Joint_update_torque_fbk(Joint* o, CTRL_UNITS trq_fbk)
{
    o->trq_fbk = trq_fbk;
    
    WatchDog_rearm(&o->trq_fbk_wdog);
}
```

Please, @valegagge, feel free to correct me if I made some mistakes or if you want to add further details.

As of now, in order to use the torque control mode we need to run the `yarprobotinterface` on a robot with `wholebodydynamics` enabled.

cc @isorrentino @pattacini @Nicogene @marcoaccame 

[1]: https://github.com/robotology/icub-firmware/issues/396#issuecomment-1768958542

marcoaccame, 2023-10-26:
> As of now, in order to use the torque control mode we need to run the `yarprobotinterface` on a robot with `wholebodydynamics` enabled.

as it is meant to be: if the board does not receives torque values cannot work in that mode. and it is `wholebodydynamics`  that sends them

valegagge, 2023-10-27:
Hi @sgiraz ,
you are completely right. You find exactly the watchdog of we discussed.

So, currently, you need `wholebodydynamics` to put the robot in torque control mode

sgiraz, 2023-11-07:
Thanks for your confirmation @valegagge.

@isorrentino, I'm pretty sure that `ergoCub1.0 S/N:000` should be available on the following dates:
- 10/11 (whole day)
- 13/11 (whole day)

Please feel free to book the robot to test the new feature you requested. Let me know if you need any help.

cc @Nicogene

isorrentino, 2023-11-09:
I've booked the robot for Monday the 13th, from 2 pm to 5 pm. @sgiraz. if you are available let's run the tests together so that you are already there in case of any issues or questions. Thanks.

isorrentino, 2023-11-10:
I had an offline discussion with @sgiraz. Unfortunately, he won't be available on Monday. Nevertheless, I plan to proceed with the tests independently. I've asked @sgiraz to summarize the required robot configuration details (YARP, icub-main versions, and configuration file modifications). This time I'm going to flash all the boards of `ergoCubSN000` to launch 'whole-body-dynamics' with the main`yarprobotinterface`. In this way, we have access to the streaming of the estimated joint torques required to enable torque control. With torque control enabled, I can test the features required in this ticket.

isorrentino, 2023-11-13:
Branches to use per each repo:

- https://github.com/robotology/yarp.git (master)
- [sgiraz/icub-firmware-shared at devel (github.com)](https://github.com/sgiraz/icub-firmware-shared/tree/devel)
- [sgiraz/icub-main at velocityThres (github.com)](https://github.com/sgiraz/icub-main/tree/velocityThres)
- [sgiraz/icub-firmware-build at devel (github.com)](https://github.com/sgiraz/icub-firmware-build/tree/devel)
- [sgiraz/robots-configuration at devel (github.com)](https://github.com/sgiraz/robots-configuration/tree/devel)

With this configuration I get the following error when I run the `yarprobotinterface`:

```shell
[INFO] EthResource::askBoardVersion() found BOARD head-eb20-j0_1 @ IP 10.0.1.20 of type mc4plus with FW = ver 3.75 built on 2023 Jul 31 11:30
[ERROR] EthResource::verifyEPprotocol() for ep = eoprot_endpoint_motioncontrol detected: pc104.version.minor = 26 and board.version.minor = 25
[ERROR] EthResource::verifyEPprotocol() detected mismatching protocol version.minor in BOARD head-eb20-j0_1 with IP 10.0.1.20  for eoprot_endpoint_motioncontrol : annot proceed any further
[ERROR] ACTION REQUIRED: BOARD head-eb20-j0_1 with IP 10.0.1.20 needs a FW update to offer services for eoprot_endpoint_motioncontrol
[ERROR] embObjMotionControl: failed verifyEPprotocol. Cannot continue!
[ERROR] |yarp.dev.PolyDriver|head-eb20-j0_1-mc| Driver <embObjMotionControl> was found but could not open

```

cc @pattacini @sgiraz

sgiraz, 2023-11-14:
I've booked `iCubGenova11` for next Friday morning (17th November). Actually, @isorrentino cannot guarantee the presence, but I'll try to do my best to complete the test successfully.

sgiraz, 2023-11-20:
Today @isorrentino and I tested successfully both the new SW and FW on `iCubGenova11`.

[image]

As you can see in the last row of the PID table, there is the new `velocityThres` parameter which makes the friction compensation tunable at runtime.

>[!Note]
>Although some quick friction compensation tests provided unexpected results, it's important to consider that these tests differed from previous tests conducted by @isorrentino on `ergoCub1.0 S/N:000` in terms of the joint tested and the robot used.

cc @Nicogene

pattacini, 2023-11-21:
Well done!
We can now proceed further with upgrading `robots-configuration` files of our robots to get them aligned with this development.

sgiraz, 2023-11-24:
As agreed during the review meeting, I'm going to open a PR on robots-configuration in order to apply the changes as described in https://github.com/robotology/icub-firmware/issues/396#issuecomment-1697136151 next week.

cc @Nicogene @pattacini

sgiraz, 2023-12-07:
PR is open:
- https://github.com/robotology/robots-configuration/pull/599

cc @Nicogene @pattacini @isorrentino

closed 2023-12-20 by pattacini