robotology/icub-firmware #409: Use the new jointset constraint `ergocubwrist`in the embedded motor control

valegagge, 2023-08-24:
Starting from the [new fw generated](https://github.com/robotology/icub-firmware-models/issues/45#issuecomment-1651338879) by the [new model](https://github.com/robotology/icub-firmware-models/pull/56), we want to use the runtime configuration instead of the macro define.

We new parameters are defined in [this special branch](https://github.com/valegagge/icub-firmware-shared/tree/feat/wristMK2_1).

**Dod**
The embedded motor control has been updated. 
The test will be carried on in a separate issue together with other specific tests.

valegagge, 2023-08-24:
I updated the embedded MC with the constraint related to the wrist as required.

@ale-git please check my code, that I pushed on your branch [here](https://github.com/ale-git/icub-firmware/commit/b2cc2467d7879d8325d1279a44b0effddfe613fc).

Thanks

Currently, I haven't implemented a strong check of the parameter of the constraint, but for now, I think the entire pipeline should work fine.
I'll do it asap.

valegagge, 2023-08-25:
Hi @ale-git,

with the `erogocub-wrist` constraint is possible to remove the macro `WRIST_MK2` ?
 
We need to remember that on the wrist there is the AMC board. So we need to use its Keil project.

ale-git, 2023-08-25:
Hi @valegagge, your last commit seems fine to me. 
About the WRIST_MK2 macro: yes, we can remove it using the constraint in its place, we only have to check that the constraint is assigned before the first #ifdef WRIST_MK2 is hit by the code flow. I can do it.

valegagge, 2023-08-28:
In agreement with @ale-git, I'll proceed to remove the macro `WRIST_MK2` if it is possible.

valegagge, 2023-08-29:
I verified that is possible to use the joint-set constraint because it is sent during the activation of the MC service.

valegagge, 2023-08-29:
Hi @ale-git ,
I removed the macro `WRIST_MK2` from the code and used the runtime configuration of the jointset (erogocubwrist).

Please have a look to the [code](https://github.com/ale-git/icub-firmware/commit/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f).

Thanks!

valegagge, 2023-08-29:
I increased the estimation since we had to remove the macro compilation.

ale-git, 2023-08-31:
Hi @valegagge I've found a bug here:

https://github.com/ale-git/icub-firmware/blob/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L702

it must be `cmd` instead of `eomc_controlmode_cmd_direct`

ale-git, 2023-08-31:
The other parts seems fine to me. About your questions in the comments:

https://github.com/ale-git/icub-firmware/blob/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L504

yes, it is related only to the wrist, we can add some suffix

https://github.com/ale-git/icub-firmware/blob/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L108

I don't  think it is necessary since they are initialized later if needed

ale-git, 2023-08-31:
I didn't find anything wrong apart from the bug reported above, I think we can give it a try as soon as the setup is ready.

valegagge, 2023-09-01:
> Hi @valegagge I've found a bug here:
> 
> https://github.com/ale-git/icub-firmware/blob/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L702
> 
> it must be `cmd` instead of `eomc_controlmode_cmd_direct`

Good catch! thanks!!!

valegagge, 2023-09-01:
Hi @ale-git,

I didn't find the commit related to the update of the amc project file. Have you already updated it?

valegagge, 2023-09-01:
> The other parts seems fine to me. About your questions in the comments:
> 
> https://github.com/ale-git/icub-firmware/blob/32ce83c9fb6665e5c3c1f81711158ff59efc3c2f/emBODY/eBcode/arch-arm/embobj/plus/mc/JointSet.c#L504
> 
> yes, it is related only to the wrist, we can add some suffix
> 

I updated the code [here](https://github.com/ale-git/icub-firmware/commit/dd169cdf70e59bbb2f4d28604a9418c4b36afda6).

ale-git, 2023-09-01:
I've updated the amc project to work with the new wrist_decoupler module, runtime configurable.

https://github.com/ale-git/icub-firmware/tree/devel

https://github.com/ale-git/icub-firmware/commit/665c37513482dc8489d9381a3e5c9cccbce1f15d

valegagge, 2023-09-04:
Dod satisfied. Closing...

closed 2023-09-04 by valegagge