robotology/icub-firmware #418: The motor temperature are sent by  2FOC  and propagated by EMS toward the YRI

valegagge, 2023-09-27:
As per title.
We need to define a new CAN periodic message and its period, but the ETH protocol is fine.
The 2FOC and EMS fw should be updated according. 

**DoD**
The temperature values are stored in the EmbObjMotionControl

valegagge, 2023-10-06:
How to develop on 2foc: [IDE](https://www.microchip.com/en-us/tools-resources/develop/mplab-x-ide) and [compiler](https://www.microchip.com/en-us/tools-resources/develop/mplab-xc-compilers)

valegagge, 2023-10-06:
Regarding the message containing the temperature value, I suggest using the `ADDTITIONAL_STATUS`.

Currently, `ADDTITIONAL_STATUS` message contains only the `interaction_mode` in the first byte. 

In the `canBusMotionControl devices` I noticed that the parser gets the `interaction_mode` without any check on the message length. see here: https://github.com/robotology/icub-main/blob/bd74c24d0aae5918d62a97ad3354ed5793a0b0f7/src/libraries/icubmod/canBusMotionControl/CanBusMotionControl.cpp#L3388

In the firmware: 
 - ems: get the `interaction_mode`. We need to update [this function ](https://github.com/robotology/icub-firmware/blob/c5c9e62515d2c2f4bf7fba98c82d9e518cbda122/emBODY/eBcode/arch-arm/embobj/plus/can/config-can-protocol/EoCANprotMCperiodic.c#L458) in order to verify that if size ==3, than we read the temperature value
 - boards with motorControllerDsp56f807: they send the message and don't read it. see [here ](https://github.com/robotology/icub-firmware/blob/c5c9e62515d2c2f4bf7fba98c82d9e518cbda122/motorControllerDsp56f807/common_source_code/can_interface.c#L1090)and [here](https://github.com/robotology/icub-firmware/blob/c5c9e62515d2c2f4bf7fba98c82d9e518cbda122/motorControllerDsp56f807/common_source_code/can_interface.c#L966)

MSECode, 2023-10-12:
Update regarding the parsing of the CAN frames:

Here the section of the code that does the parsing of the message sent by the 2FOC code:

[image]

As you can see, we are parsing the CAN frames using the `for loop` over the only 2 joints that the FOC can control. 
Therefore, we are taking the third and fourth bytes of the frame and saving to the temperature of the first motor entity. Then the fifth and sixth bytes (being the temperature a typedef int16_t) are saved to the temperature of the second motor entity.

MSECode, 2023-10-12:
Done.
Remains only to define the conversion factor from V diff to Celsius

MSECode, 2023-10-24:
The activity has been completed and tested.
The temperature values are stored in the `embObjMotionControl` class in this array: `double *_temperatureValues` and a yarp module for retrieving the motor temperature and sending it to a user-defined yarp port has been developed and published here: https://github.com/MSECode/motionControlModule/tree/feature/temperatureManager.

It is possible to check the output of the test in [this issue](https://github.com/icub-tech-iit/five/issues/175)

valegagge, 2023-10-27:
Dod satisfied. closing

closed 2023-10-27 by valegagge