robotology/icub-firmware #427: Fix bug on 2foc ver 3.6 related to the encoder readings

valegagge, 2023-10-17:
On ergocub 1.1 we experienced a block of the motor of the right shoulder pitch when it goes down with high velocity.

ASAP I put here the reference

**Dod**
bug fixed

maggia80, 2023-10-18:
today @AntonioConsilvio tested the legs with the same firmware and he managed to replicate the same issue on the `left or right ankle roll`

valegagge, 2023-10-19:
After a chat with  @AntonioConsilvio, he noticed that the current consumed is around 12A (read in the yarpmotorgui). He stopped the joint before it could go into fault. 
In order to exit from this situation, he had to move by hand the joint of about 10 deg.

ale-git, 2023-11-03:
The high current depends - very likely - on the loss of rotor angle measurement: if the motor torque drops because of lack of synchronization between rotor and stator field, the controller increases the current without effectiveness.

ale-git, 2023-11-03:
Comparing the 3.3 and 3.6 firmware I noticed that there is a difference in the encoder index signal front reading, that is configurable.

[image]

The IMV bits have always been set to `11` for iCub ROIE motor encoders but, while testing the new ErgoCub joint with MRIE encoders, I noticed that some index detection were missing when the motor was rotating clockwise and thus, after some tries, I found that the problem was solved by changing the configuration to `10`.
So: in the 3.3 firmware IMV="11", while in the 3.6 firmware IMV="10".

Now, we see that the 3.6 firmware has problems with the lecture of the index in one direction with the old ROIE encoders, so I suspect that it is due to the IMV configuration made for MRIE encoders. I prepared a 3.6 firmware with IMV="11" in order to check if it solves the issue. In this case we have to make the IMV control bits configrable at runtime. The motor configuration CAN message use one byte for the number of polar couples of the motor that seems eccessive, so we can change the protocol so that 2 bits are used for IMV configuration and 6 bits are used for polar couples (at present the maximum number of polar couples is 20 and it is unlike that it can excess 62).

maggia80, 2023-11-03:
MRIE is equipped with AEDR-9830

[image]

The index phase in the MRIE is set in the board and it's 180:  

[image]

and 
[image]

LCORE with AEDR-8501-102

[image]

Index phase for the LCORE is set in the chip and it's 180 degree ->
[image]

maggia80, 2023-11-03:
the connectors have same pinout: 

MRIE connector is:

[image]

and LCORE:

[image]

maggia80, 2023-11-03:
> Comparing the 3.3 and 3.6 firmware I noticed that there is a difference in the encoder index signal front reading, that is configurable.
>
> [image]
>
> The IMV bits have always been set to `11` for iCub [...]

MRIE has the index associated to CHB, while LCORE to CHA. That's why you need to change the IMV register

maggia80, 2023-11-03:
however, having the phase to 180 degrees, there is something that does not sound to me. With 180 the state of phase A and B when the Quadrature is x4 is not defined!

In case of the MRIE, CHA has a transaction from 1 to 0, while CHB is fixed to 1.
for LCORE, CHA is fixed to 1, while CHB has a transaction from 0 to 1.

ale-git, 2023-11-03:
> however, having the phase to 180 degrees, there is something that does not sound to me. With 180 the state of phase A and B when the Quadrature is x4 is not defined!
>
> In case of the MRIE, CHA has a [...]

I don't understand this part, maybe you can explain it to me face to face next week at CRIS?

maggia80, 2023-11-03:
this is the case of the LCORE [image]

maggia80, 2023-11-03:
and this is the MRIE: 
[image]

ale-git, 2023-11-03:
This is the 3.3 version with MRIE Index configuration.
[2FOC-V3.3.MRIE.zip](https://github.com/robotology/icub-firmware/files/13254325/2FOC-V3.3.MRIE.zip)

valegagge, 2023-11-05:
Hi @ale-git ,
this isn't the correct issue where attach the exe. please see this [comment](https://github.com/icub-tech-iit/ergocub-design-joint/issues/179#issuecomment-1792835172).

Anyway I already moved the .zip file. 😸

valegagge, 2023-11-10:
We decided do not to use the 2foc fw ver3.6

closed 2023-11-10 by valegagge