robotology/icub-firmware #428: joint controlled with AMO boards used as incremental encoder stop reading going to a HW limit

maggia80, 2023-10-18:
## Description

We noticed it on the left pitch and roll shoulder in ergoCub1.1 S/N:001. The two joints are controlled in position and after a while even if there aren't any position commands the joint goes toward an HW limit. While moving the AMO reading seems to be blocked. 
Sometimes the joint doesn't reach the HW limit, but it remains quite far from the previous position while reading a similar angular position.     

##Hypotesis 

The AMO stops reading for some reason and the PID to reduce the error from the setpoint and the current position generates a PWM that moves the joint trying to reduce the error, but since the AMO is stuck to a fixed value, the error remains constant and the motor keeps moving. 

Possible reasons:
1) the reading of the AMO is interpreted as a spike and the value is kept constant for a long period of time.
2) the AMO is not reading for some cable issues
3) the position of the AMO is too far or not well aligned with the magnetic target causing spikes.

maggia80, 2023-10-18:
cc: @antonioconsilvio, @ale-git @valegagge

maggia80, 2023-10-18:
@antonioconsilvio would be very helpful if you could log the encoder and motor position while this phenomenon occurs.

MSECode, 2025-01-15:
After all the analysis done and collected in this epic: https://github.com/icub-tech-iit/study-encoders/issues/21 we can summing up all the solutions found and the problems that are still open in the following points:
- the problem there were during the calibration have been partially solved. We have understood why sometimes the joint was failing to reach the hard-stop position and now that is solved. 
- for the problem related to the impossibility to reach sometimes the startup position. This has been partially solved and we have almost identified the core of the problem. Anyways, with the improvements added to the code that is really unlikely to happen
- for the problem of the AMO that sometimes reads random values. This problem is still not 100% clear since we were able to reproduce that behavior only by moving the AMO encoder board far from the magnetic disk while the joint is moving but this condition should never happen and looking at the mechanics of the joint that cannot happen when the robot is used properly (if that happen it means that the joint is actually broken)
- if we have problem on the cabling such as one of the signal cables is broken or detached or we disconnect the AMO encoder connector the joint is SET in HW FAULT and it stops
- if the AMO encoder is not fixed at its standard position or if the magnetic disk is damaged the fw will send on the upper level plenty of warnings about reading of `spikes`. Since that warning is related to different erroneous conditions we have decided to improve the AMO diagnostic so that the users are more aware of what is actually happening when the joint is not working fine

cc: @valegagge