robotology/icub-firmware #435: Implement hard stop calibration, AMO-less operation in Ergocub

ale-git, 2023-11-03:
At present the Ergocub and iCub 3 robots use AMO angular position sensors as incremental encoders as a workaround to avoid the absolute position loss problem. This requires hard stop calibration that has been implemented some time ago. Now we want be able to run a joint without the AMO sensor, only relying onto the motor encoder, in case of complete AMO sensor breakage. Some modifications are thus required to synchronize the EMS and the 2FOC boards at startup, in this order:
1) the 2FOC runs its own calibration (encoder Index calibration, current sensor calibration)
2) the EMS performs hard stop calibration of the joint

Definition of Done:

 New EMS firmware tested and merged.

@valegagge

valegagge, 2023-11-06:
Hi @ale-git,
to achieve this goal we don't need to change the 2foc fw, do we?

ale-git, 2023-11-06:
No, it has no dependecy.

ale-git, 2023-11-08:
I've tested the fw on Ergocub 1.1 and it works properly.

valegagge, 2023-11-10:
PR opened here: https://github.com/robotology/icub-firmware/pull/438.

valegagge, 2023-11-15:
## add this feature in the new sw-distro: 
[image]

valegagge, 2023-11-29:
Dod satified. Closing

closed 2023-11-29 by valegagge