robotology/icub-firmware #446: Validate the q_enc reading improvement on the ergocub 1.X joints

valegagge, 2023-12-07:
# Task
During the test on ergocub 2.0 joints, we noticed that the index interrupt service routine can be executed with a big delay that is used to disrupt the FOC control at motor high speed. So we changed the position counter reset policy from software (on interrupt) to hardware (managed by the QE peripheral of the dspic33f). 

This change solved the problem on the ergocub joints 2.0.
Now we want to perform the same test on the ergocub 1.x joints

# Dod
- [ ] Test done
- [ ] Data collected and reported here. 
- [ ] Collect the same data on ergocub 2.0 and put here the report.

For collecting data, please perform the same tests [here](https://github.com/icub-tech-iit/ergocub-design-joint/issues/179#issuecomment-1802128164).

valegagge, 2023-12-13:
Here I report the tests done on the **small joint on ergocub 2.0** highlighting the difference between running the joint with the official fw version and with the bug fixed.
[image]

valegagge, 2023-12-13:
Here I report the tests done on the **torso pitch joint of ergocub 1.x** highlighting the difference between running the joint with the official fw version and with the bug fixed.

[image]

valegagge, 2023-12-14:
cc @maggia80

maggia80, 2023-12-14:
These are amazing results!!! This means a big improvement in the performance/efficiency of our robots!

valegagge, 2023-12-19:
dod satisfied. Closing

closed 2023-12-19 by valegagge