robotology/icub-firmware #450: Make the environment applications of ETH boards compilable with the last Keil compiler

valegagge, 2023-12-18:
Currently, we need to use the compiler version 5 to compile the EMS updater.

We need to update the project to use the last compiler.

**Dod**
The environment applications of ETH boards compile with the last Keil compiler

valegagge, 2024-02-08:
see also https://github.com/robotology/icub-firmware/issues/465

marcoaccame, 2024-02-12:
Hi @valegagge, if you want to add a target V6 to the `eUpdater` of the `ems`, in [here](https://github.com/robotology/icub-firmware/tree/master/emBODY/eBcode/arch-arm/board/ems004/env/eUpdater-embot/proj) is a test project that uses the compiler V6. It was a working project for the porting of the `eUpdater` on the `amc`.

It is easy to do (I attach the modified project), I just wonder if we need to do it and release a new binary.
 
[eom-eupdater-open-v6.zip](https://github.com/robotology/icub-firmware/files/14237635/eom-eupdater-open.zip)

valegagge, 2024-02-12:
This isn't my wish 😸 , but I guess we need to do it at some point.
So if there is an imminent necessity we can schedule it.

marcoaccame, 2025-10-06:
closing because of:
- https://github.com/robotology/icub-firmware/pull/638