robotology/icub-firmware #451: Develop the diagnostic related to the q_enc motor reading in the 2 foc

valegagge, 2023-12-19:
After validating the fix of q_enc reading bug, we want to develop a clear diagnostic that helps the user understand when e which type of error we can have on the q_enc readings.

Dod
- [ ] definition of the diagnostic
- [ ] implementation 
- [ ] tests

valegagge, 2023-12-19:
The tests will be addressed in #452

valegagge, 2023-12-27:
Hi @ale-git, 
I know you already developed the required feature. 🚀 
Please add here the link to the branch of your code and some info related to the diagnostic.

Then you can put it in review

ale-git, 2023-12-28:
This is the draft PR.

https://github.com/robotology/icub-firmware/pull/455

ale-git, 2024-01-08:
The diagnostics is separated in two types of fault detection:
- broken index
- encoder dirty.

Notice: the "phase_broken" error type can't be detected in motors without Hall sensors, because the motor stops if the encoder stops counting. The "phase_broken" error is still detected, but only in motors equipped with quadrature encoders and Hall sensors.

This is the new diagnostics - pretty simple - moved from the interrupt service routine to the encoder reading function called at 40 kHz. 
In order to detect the dirty encoder disk it compares the expected values of the encoder (within some tolerance) and the real value when the index pin is high. In order to detect the broken index it checks the encoder value (within some tolerance) for out of range.

`
inline int QEgetPos()
{
    int poscnt = (int)POSCNT;
    
    if (QE_USE_INDEX)
    {
        static int poscnt_old = 0;

        if (qe_index_found)
        {
            if (QEICONbits.INDX)
            {
                if (poscnt_old>QE_ERR_THR && poscnt_old<QE_RESOLUTION-QE_ERR_THR)
                {
                    QE_RISE_ERROR(dirty);
                }
            }
            else if (poscnt<-16 || poscnt > QE_RESOLUTION+16)
            {
                QE_RISE_ERROR(index_broken);
            }
        }
        
        poscnt_old = poscnt;
    }

    return __builtin_divsd(((long)poscnt)<<16,QE_RESOLUTION);
}
`

https://github.com/ale-git/icub-firmware/blob/3cd6e349a5d096758819f04392b849e8e88da8b2/emBODY/eBcode/arch-dspic/board/2foc/appl/2FOC-V3/src/qep.c#L131

valegagge, 2024-01-08:
Dod satified. The test will be performed in #452 .

closing

closed 2024-01-08 by valegagge