robotology/icub-firmware #454: Improve the CAN monitor configuration inside the `embot_app_eth_theBATservice`

valegagge, 2023-12-27:
As explained here https://github.com/robotology/icub-firmware/issues/445#issuecomment-1870195848, we need to use a default configuration if the sent configuration is empty.

**Dod**

- [ ] the above mentioned check has been implemented
- [ ] the entire feature is tested on the robot.

MSECode, 2023-12-28:
Some of the improvements have been done. Specifically we have that:
- update the battery configuration files setting the CAN address to `2` and the PR is ready here: https://github.com/robotology/robots-configuration/pull/603
- in the firmware we have added the check on the `canmonitor` checkPeriod and if equal to zero, meaning not set, we are setting it to `250` so that it is lower than the max value, which is 254 being a uint8 datatype, and greater than the transmission rate of the bat which is 100 ms. Here the PR with the changes: https://github.com/robotology/icub-firmware/pull/447

MSECode, 2024-01-04:
Tested current improvements pushed to the branches linked above on ergoCub robot and here the screen:
[image]

MSECode, 2024-01-04:
As you can see the BAT is sending correctly the messages when the YRI is started and no weird error is streamed on the YRI. (A couple of restarts have been done) and setting the check of the canmonitor at 250ms we have no error related to it continuosly printed on the YRI.

MSECode, 2024-01-05:
CAN Frames and stop and start transmission message between frames

[image]

MSECode, 2024-01-09:
All the PRs are here:
- [icub-firmware](https://github.com/robotology/icub-firmware/pull/447)
- [icub-firmware-shared](https://github.com/robotology/icub-firmware-shared/pull/91)
- [icub-firmware-build](https://github.com/robotology/icub-firmware-build/pull/126)

MSECode, 2024-01-09:
cc: @valegagge

valegagge, 2024-01-09:
it is also necessary the PR https://github.com/robotology/robots-configuration/pull/603

MSECode, 2024-01-10:
PR closed --> moving to close state

closed 2024-01-10 by MSECode