robotology/icub-firmware #473: Motor in HF due to overheating with temperature very low

isorrentino, 2024-03-01:
On the ergoCubSN001 boards there is a test firmware to read the motor temperatures. While running experiments, even if the temperature of the `l_ankle_roll` was low, the motor went into overheating and I had to restart the robot.

Here are some plots.

1) Before the overheating
[image]

2) When in overheating
[image]

[image]

3) When restarted the robot right after the HF
[image]

Here is the log:
[log_ergocub-torso_yarprobotinterface_2117_temperature.txt](https://github.com/robotology/icub-firmware/files/14459696/log_ergocub-torso_yarprobotinterface_2117_temperature.txt)

cc @valegagge @DanielePucci @maggia80 @GiulioRomualdi

DanielePucci, 2024-03-01:
This is pretty important to be fixed since if it happens during the demo the robot falls when it should not 

@MSECode

MSECode, 2024-03-01:
As you can see from image2 and from what written in the documentation here, where all errors are detailed, https://icub-tech-iit.github.io/documentation/temperature_sensors/software/dataflow/#error-handling

the problem is basically due to the fact that the related ems board was unable to get any reading from the TDB for more than 20 seconds as you would have the cable broken or maybe constantly squeezed on a section for that time. Regarding the error message we have already planned to update the Diagnostic with a specific error message but anyways you have the full documentation explaining everything. 
So I suggest you to check the cable. Looking at the data above, I would say that there's not any fw/sw problem but more likely a hw problem with cable position or connection on the connector.

DanielePucci, 2024-03-03:
Thanks @MSECode. Anyways, concerning 

> the problem is basically due to the fact that the related ems board was unable to get any reading from the TDB for more than 20 seconds as you would have the cable broken or maybe constantly squeezed on a section for that time. [...]
> So I suggest you to check the cable. Looking at the data above, I would say that there's not any fw/sw problem but more likely a hw problem with cable position or connection on the connector.

The joint shall not switch to HF if there is a hardware problem that alters the temperature reading since the risk of making the robot fall because we have a problem in the temperature reading is definitely too high.

In the avionics standards, faulty measurements lead to deactivating the autopilot and leave the process of decision making to higher control levels (sometimes the pilots) [1,2] while keeping the same values of thrust and attitude controls. So, to avoid an aircraft [loss-of-control](https://en.wikipedia.org/wiki/Loss_of_control_(aeronautics)), the low level control system **does not shut down the aircraft motors in the case faulty measurements**: the principle of **do nothing** rules in avionics [3]. So, it would be nice to have a similar paradigm on the robot.

[1] https://youtu.be/LA-WeacusP4?t=1385
[2] https://skybrary.aero/articles/unreliable-airspeed-indications
[3] https://www.youtube.com/watch?v=LA-WeacusP4&t=1414s&ab_channel=S.Babey

DanielePucci, 2024-03-04:
Available for a meeting @MSECode

valegagge, 2024-03-04:
Hi @DanielePucci,
I understand your point of view. It is clear, but we need to be sure that the temperature doesn't exceed the max value, and therefore is important to have the temperature value. In case of error readings for a long time, we need to disable the motor, but, obviously, during the demo, this behavior is not acceptable. For this, you can disable temperature check during a demo. (I know it is annoying)

In the case of @isorrentino experiment, it is important to understand what was going on. If it is an HW problem, as we suspect, we have to fix it.

So, I'd like to know:
- is the problem reproducible?
- does this issue happen only during @isorrentino experiment? It seems strange that it happens by changing the configuration files only.

Thanks heap!!

isorrentino, 2024-03-04:
> does this issue happen only during @isorrentino experiment? It seems strange that it happens by changing the configuration files only.

After my experiments, Evelyn wasn't able to run experiments due to https://github.com/robotology/icub-tech-support/issues/1763#issuecomment-1976357540. I don't know whether other guys used the robot after that (this morning).

valegagge, 2024-03-04:
> > does this issue happen only during @isorrentino experiment? It seems strange that it happens by changing the configuration files only.
> 
> After my experiments, Evelyn wasn't able to run experiments due to [robotology/icub-tech-support#1763 (comment)](https://github.com/robotology/icub-tech-support/issues/1763#issuecomment-1976357540). I don't know whether other guys used the robot after that (this morning).

Thanks @isorrentino. Please contact us when you'll face up with the issue again, checking if the problem is reproducible and which is the temperature value.

Thanks again

DanielePucci, 2024-03-04:
Hi @valegagge, thanks! Concerning
> In case of reading erros for a long time, we need to disable the motor, but, obviously, during the demo, this behavior is not acceptable. For this, you can disable temperature check during a demo. (I know it is annoying)

Disabling the temperature reading files one by one is error prone - we may really not remember it - so it would be better to avoid it. 

Furthermore, we would like to work on temperature dependent parameters, so disabling the temperature reading during the demos would render ineffective the temperature dependency whose aim is indeed to robustify the demos.

isorrentino, 2024-03-06:
Today, it happened again. Some minutes before the temperature of the right knee was 

[image]

and after some minutes the joint went overheating
 
[image]
[image]

and we had to restart the robot.

Also, this time we were not running experiments when it happened so we do not have data between the two plots I dropped here. In addition, the `yarp logger` was saturated and we do not have the error from the log.

cc @valegagge @MSECode @DanielePucci @GiulioRomualdi

valegagge, 2024-06-07:
Hi @isorrentino,

@MSECode and I performed several tests and we concluded that the issues you spotted are due to electromagnetic noise that disturbs motor temperature readings.

We know this problem annoys you while doing the experiments, but we are working to find an HW solution.

Therefore, I suggest closing this issue. Ok?
 I'll keep yu updated with the HW solutions.

closed 2024-06-18 by valegagge