robotology/icub-firmware #476: Improve battery (BAT) data streaming and management

MSECode, 2024-03-06:
## Task description :memo:
As requested and discussed in https://github.com/icub-tech-iit/tickets/issues/3343 some improvements should be developed for BAT board, specifically:
- currents streaming
- voltage streaming when connected to either batteries or power supply
- avoid spamming the terminals if the battery is off (we noticed that some warnings are thrown in the terminal when the BMS is off)

## Definition of Done :white_check_mark:
New fw tested and working as AMI group desires

MSECode, 2024-03-06:
Working branches are the following:
- https://github.com/MSECode/icub-firmware-shared/tree/feature/batUpdate
- https://github.com/MSECode/icub-main/tree/feature/batUpdate
- https://github.com/MSECode/icub-firmware/tree/feature/batUpdate

MSECode, 2024-03-06:
cc: @valegagge

MSECode, 2024-03-06:
[image]

MSECode, 2024-03-06:
After F2F discussion w/ @valegagge and @maggia80 regarding multiple aspect about BAT we have decided the following:
- we can send the INPUT VOLTAGE for the BAT as the first 2 bytes of the CAN packet with id `0x620`. Even if we know that we are using same network variables for both BMS and BAT and the same CAN packet we are modifying the meaning of the values for the BAT only so that we can give to the user all the pieces of information they need when using the system connected with either the power supply or the batteries. Therefore, even if the CAN message sent by the BAT differ from the one of the BMS regarding the protocol, the user can have always all the info
- adding the total current absorbed by the system on bytes 2 and 3 of the CAN message `0x620` sent by the BAT. 
- evaluation of the Battery Charge on the BAT side should be revised and moved on the fw since the calculation is done with a frequency smaller than the streaming of its value

MSECode, 2024-03-06:
So, as detailed above the following changes have been made:
- if battery charge raw value is `zero` on the high level we are setting it to NAN, so that we can somehow avoid spamming the terminals with the law battery charge message as shown here:

[image]

- changed the lower limit for the battery charge calculation from 33000 mV to 30000 mV according to the line identified on the graph of the robot battery discharge here:

[image]

Even if not 100% accurate, especially when the SoC is low the line can be use to approximate the level of charge in the robot batteries. Now, at a first look to the values given by the setup used, which are shown here:

[image]

it seems that the battery charge percentage correlated to the instant voltage is close to the more accurate value given by the BMS. So I suppose that we can currently accept this approximation. Then, when tested on the robot we can see if the values between BAT and BMS are compatible or they differ a lot and eventually use an offset or change the order of the approximation for the curve to follow better the lower part of the discharge curve.

- finally now the total current absorbed by the system is sent up to the ems from the BAT as well as the temperature value.

Therefore, as things now, all the points requested on the ticket should be satisfied. Need to make some tests on the robot for checking the values obtained since I've done tests only on the bench setup.

maggia80, 2024-03-10:
ok!

MSECode, 2024-03-12:
I've done some tests on ergoCub, here the results:

## current status w battery switched OFF

### BMS

[image]

### BAT

[image]

## current status w battery switched ON

### BMS

[image]

### BAT

[image]

## updated status w battery switched OFF

### BMS

[image]

### BAT

[image]

## updated status w battery switched ON

### BMS

[image]

### BAT

[image]

MSECode, 2024-03-14:
I need to do another test using only the battery and disconnecting the power supply thus to cover all cases

MSECode, 2024-03-14:
yes, sure, the documentation needs to be updated. Then:

- Voltage: as you said
- Current: for BAT is total current absorbed by the system, for BMS the current in or out from the battery
- SoC: as you see is available only if battery is connected otherwise NaN and for BAT is calculated on the Voltage

maggia80, 2024-03-14:
and for BAT is calculated on the Voltage
> 
> 

Which voltage? Should be the voltage of the battery if available.

MSECode, 2024-03-14:
> the

Exactly. It is the Battery voltage as here:

`Battery_charge = 100*(mean.V_BATTERY-Battery_low)/(Battery_high-Battery_low)`

maggia80, 2024-03-15:
<html>
<body>
<h2 dir="auto" style="box-sizing: border-box; margin-top: 24px; margin-bottom: 16px; font-size: 1.5em; font-weight: var(--base-text-weight-semibold, 600); line-height: 1.25; padding-bottom: 0.3em; border-bottom: 1px solid var(--borderColor-muted, var(--color-border-muted)); color: rgb(201, 209, 217); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; background-color: rgb(13, 17, 23); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">updated status w/ battery switched OFF</h2>
BMS | BAT
-- | --
  |  

</body>
</html>

@MSECode  when the battery is switched off you should read NaN because the voltage of the pin Battery_voltage 

[image]

valegagge, 2024-03-15:
Hi @maggia80 ,
currently the BAT baord is publishing the temperature, I suppose related to the battery, but its value is very different from BMS one.
[image]

We checked the code of Battery, where it calculates that value, but it isn't clear to us. ([Here the code](https://github.com/robotology/icub-firmware/blob/ebc395c879108a8f04e82f530b91fc4cb8f78f41/emBODY/eBcode/arch-arm/board/bat/BAT_Rev_B/Src/main.c#L394). )

So the questions are:
1.  is the BAT calculating the temperature correctly?
2. If yes, do we want to show a value that differs from the correct one so much? (~ double)

valegagge, 2024-03-18:
Hi @MSECode,
hereafter the issue to address the documentation updates. https://github.com/icub-tech-iit/documentation/issues/342 as we already discussed.

MSECode, 2024-03-20:
> Hi @maggia80 , currently the BAT baord is publishing the temperature, I suppose related to the battery, but its value is very different from BMS one. [...]

I've checked the origin of that `adc_values.Temp_sensor` value. It is referred to this MACRO initialization: 
`#define LL_ADC_CHANNEL_TEMPSENSOR          (LL_ADC_CHANNEL_17 | ADC_CHANNEL_ID_INTERNAL_CH) /*!< ADC internal channel connected to Temperature sensor. On STM32L4, ADC channel available only on ADC instances: ADC1, ADC3. */` 

And it is specifically the internal temperature of the microcontroller.
Therefore, I suppose that can be omitted from the values sent towards the user and added to the CAN message `0x620` since it is not informative to the scope of that CAN frame, since it could be quite different from both the battery pack temperature and the temperature of the system itself.

cc: @maggia80

maggia80, 2024-03-20:
and for BAT is calculated on the Voltage
> 
> 

Which voltage? Should be the voltage of the battery if available. 

Yes, I agree

MSECode, 2024-03-21:
### Results after tests on `ergoCub001`

After testing the latest firmware, which is available in this PR, the results are as follows.
First of all, now all the requested data are available and they allow the user to have a good idea of the status of the system and eventually of the battery pack specifically. The images that follows explain this.

On all the images it is possible to see the following important points:

- left-hand-side tab shows the YRI logs and it is possible to see that every 10s ( or depending on the value of `embot::core::relTime periodofreport` defined in the `CANmonitor::Config defaultcanmonitorconfig` set in the fw in the file `embot_app_eth_theBATservice.cpp` (for now is hardcoded and cannot be updated from the configuration. It will be possible after the completion of https://github.com/robotology/icub-main/issues/927 ))
- middle tabs shows the values printed on the yarp port `ergocub/battery/bms/data:o` that are in order: `Battery_Voltage [V] | Current towards/outwards (+/-) the battery-pack [A] | battery-pack StateofCharge [%] | battery-pack Temperature [^C] | BMS_error_status`
- right-hand-side tab shows the values printed by the yarp port `ergocub/battery/bat/data:o` that are in order: `Input_Voltage [V] (Can be either the power-supply voltage or the battery voltage, the highest anyway) | Total current absorbed by the system (sum of hsm, cpu, motors) [A] | battery-pack StateofCharge [%] (But calculated using the Battery Voltage) | Not Used (NaN) | BAT_status`

**NOTES:**
- In some screen there's  no one of the port printed since the values are not informative because battery-pack switched off (BMS OFF) or power-supply not  connected (BAT OFF)
- BAT yarp port values are defined to give to the user an idea of the status of the whole system (i.e. the robot)
- BMS yarp port values are defined to give to the user an  idea of the status of the information related to the battery-pack
- If SoC raw value is equal to 0.0 we are setting its "yarp" value to NaN so that the terminal is not flooded with the Warning message of critical battery charge even when it is meaningless.
 
#### Values displayed on the yarp ports when we have both BMS and BAT enabled and working:

[image]

NOTES:
- Voltages are comparable
- BMS gives the current going out from the batteries (almost null since we have the power supply enabled)
- BAT is showing the total current absorbed by the system
- SoC (correct is the one of the BMS) but the one displayed in the BAT port is close to it (fine for us)

#### Values displayed on the yarp ports when we have both BMS and BAT device enabled by configuration but the battery-pack enabler switch is in OFF position:

[image]

NOTES:
- See on the logs that the canmonitor is notifying about the impossibility to find the device enabled 
- All values of the BMS in the canmonitor are 0 or NaN 
- In the BAT the SoC is NaN 

#### Values displayed on the yarp ports when we have only BAT enabled (only power-supply):

[image]

#### Values displayed on the yarp ports when we have only BMS enabled (robot is powered by battery-pack):

[image]

cc: @valegagge @maggia80

closed 2024-04-02 by MSECode